rerun-io/rerun #1315: Viewer crashes on save

AnkeAnke, 2023-02-15:
**Describe the bug**
Clicking on the `Save...` menu item or keying `ctrl+S`, the viewer crashes. 

**To Reproduce**
Steps to reproduce the behavior:
1. Log some data (via connecting from other process)
2. Try to save (`Save...` or `ctrl+S`)
3. Process crashes

**Expected behavior**
No crash, preferably even a save file.

**Backtrace**
Couldn't get a backtrace (even with `RUST_BACKTRACE=1`). But here's the terminal output:
```
(python3:8726): GLib-GObject-WARNING **: 13:42:26.828: cannot register existing type 'GdkDisplayManager'

(python3:8726): GLib-CRITICAL **: 13:42:26.828: g_once_init_leave: assertion 'result != 0' failed

(python3:8726): GLib-GObject-CRITICAL **: 13:42:26.828: g_object_new_with_properties: assertion 'G_TYPE_IS_OBJECT (object_type)' failed
Segmentation fault (core dumped)
```

**Desktop:**
- OS: Pop! OS 22.04 LTS (Ubuntu with benefits)
- Windowing system: X11
- Python: 3.10.4 (June 22, GCC 11.2)
- GLib: Ubuntu GLIBC 2.35-0ubuntu3

emilk, 2023-02-15:
> No crash, preferably even a save file.
😆

emilk, 2023-03-02:
Likely fix:
* https://github.com/rerun-io/rerun/pull/1402

closed 2023-03-27 by emilk