rerun-io/rerun #1316: Improve compile times

Wumpf, 2023-02-15:
and aspire to reduce number of dependencies and code bloat in the process.

Wumpf, 2023-02-15:
rebuild on dev with M1 max.

[image]

Quite surprised about zlib being slow here!

[cargo-timing-20230215T130337Z.html.zip](https://github.com/rerun-io/rerun/files/10743344/cargo-timing-20230215T130337Z.html.zip)

Wumpf, 2023-02-15:
very similar results on second run

Wumpf, 2023-02-17:
Getting rid of zstd is somewhat hard - it *is* just really really good it seems https://github.com/EmbarkStudios/puffin/pull/130

emilk, 2023-04-17:
I think we can switch from `zstd` to [`lz4_flex`](https://github.com/pseitz/lz4_flex). To get the benefits we must first remove `zstd` from our `puffin` dependency:

* https://github.com/EmbarkStudios/puffin/issues/131

emilk, 2023-05-13:
Current state on my M1 MacBook Pro

```
sccache --stop-server && rm -rf /Users/emilk/Library/Caches/Mozilla.sccache && sccache --start-server
cargo clean && cargo build --timings -p rerun
```

2m 12s

[image]

emilk, 2023-05-20:
This halves compilation times for users only using the `rerun` crate as a logging library:
* https://github.com/rerun-io/rerun/pull/2177

emilk, 2023-12-11:
Things has improved - down to 87.0s now for `cargo clean && cargo build --timings -p rerun` on my M1 MacBook Pro

[image]

Some things we could maybe cut out with some effort:
* `tracing-attributes` - 23.4s cpu time - we don't use it, but some dependencies do
* `regex` - 9s cpu time - a bunch of dependencies use regex for minor things, but should switch to `regex-lite`
* `tokio-macros` - 11s of CPU time for `#[tokio::main]`?

Wumpf, 2024-06-18:
Status update: On c5f817686c6a1b8723ddeee25a9aed0ff4fa1e92 (somewhere between 0.16 and 0.17 ;)

M1max  MacBook Pro (very similar to emils machine above)
66.5s and 336 dependencies 

[image]

`re_types` is the biggest unit with quite some distance

Wumpf, 2025-02-12:
we since then doubled our number of crates unfortunately. Due to the new grpc dependencies we're at 449 crates for the SDK alone 😞

Build times on my windows machine (!, not comparable with previous posts) via `cargo clean && cargo build --timings -p rerun` are at 1m 38s.

html file: (remove .txt - github doesn't allow htm)
[Cargo Build Timings — rerun 0.23.0-alpha.1+dev.htm.txt](https://github.com/user-attachments/files/18767932/Cargo.Build.Timings.rerun.0.23.0-alpha.1%2Bdev.htm.txt)

emilk, 2025-02-24:
For simplifying `re_types`, these will help:
* #6388 
* #6819

osiewicz, 2026-10-13:
I'm interested in looking into this as a bystander. All build invocations in this thread so far include `cargo clean`; do you care about rebuild times after making edits to a set of particular crates? Do you have any goal in mind for the target DX?

Wumpf, 2026-10-23:
It's a very very outdated ticket and we've done a lot to curb iterative compile times. Some of it quite recent. But the ticket doesn't make sense anymore in this form.

closed 2026-10-23 by Wumpf