rerun-io/rerun #1317: Warning on every resize

AnkeAnke, 2023-02-15:
**Describe the bug**
When resizing the viewer window, a warning is logged every time, spamming the terminal.
Also logged that warning on first ever startup (not subsequent ones).

**To Reproduce**
Steps to reproduce the behavior:
1. Open viewer
2. Resize (drag the edge)
Does not happen when switching fullscreen/windowed.

**Expected behavior**
No incessant warnings.

**Warning**
```
2023-02-15T13:13:17.777258Z  WARN wgpu_core::device: Requested size 1704x1498 is outside of the supported range: Extent3d { width: 2113, height: 1498, depth_or_array_layers: 1 }..=Extent3d { width: 2113, height: 1498, depth_or_array_layers: 1 }
```
Requested size changes of course.

**Desktop (please complete the following information):**
- OS: Pop! OS 22.04 LTS (Ubuntu with benefits)
- Windowing system: X11
- Python: 3.10.4 (June 22, GCC 11.2)
- Screen: 4K (3840x2160)

**Additional Information**
Previous output (on every startup):
```
2023-02-15T12:35:37.579295Z  INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 2    
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

2023-02-15T12:35:37.657051Z  WARN wgpu_hal::vulkan::instance: Disabling presentation on 'Intel(R) UHD Graphics 630 (CFL GT2)' (id 0x55774fb35fb0) because of NV Optimus (on Linux)   
```

teh-cmc, 2023-02-15:
Can you try and see..:
- ..what happens when you force Vulkan to use the internal GPU?
E.g. `export VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/intel.json`  then run Rerun as usual
- ..what happens when you force Vulkan to use the internal GPU?
E.g. `export VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/nvidia.json`  then run Rerun as usual

In both cases, having the full startup and resize logs would be very useful (hopefully) :pray: 

(Those ICD paths may not match exactly what's on your system, should be close tho)

AnkeAnke, 2023-02-16:
Here's the outputs for all my (working) drivers:

```
> VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/intel_icd.x86_64.json python3 -m rerun
[time]  INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 2    
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0
[time]  WARN wgpu_core::device: Requested size 2877x1498 is outside of the supported range: Extent3d { width: 2545, height: 1498, depth_or_array_layers: 1 }..=Extent3d { width: 2545, height: 1498, depth_or_array_layers: 1 }
```

```
>VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/nvidia_icd.json python3 -m rerun
[time] INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 2    
[time] WARN wgpu_core::device: Requested size 3643x1401 is outside of the supported range: Extent3d { width: 3646, height: 1401, depth_or_array_layers: 1 }..=Extent3d { width: 3646, height: 1401, depth_or_array_layers: 1 }
```
In both cases, the resize warning does not happen on startup, but resize only.

And just for fun:
```
> VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/lvp_icd.x86_64.json python3 -m rerun
[time] INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 2    
[time] WARN wgpu_core::instance: Missing downlevel flags: SURFACE_VIEW_FORMATS
The underlying API or device in use does not support enough features to be a fully compliant implementation of WebGPU. A subset of the features can still be used. If you are running this program on native and not in a browser and wish to limit the features you use to the supported subset, call Adapter::downlevel_properties or Device::downlevel_properties to get a listing of the features the current platform supports.    
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.
[time] WARN wgpu_hal::vulkan: Suboptimal present of frame 0
```
Lavapipe actually works (though slow as hell) and does not throw the resize warning!
So I guess I now at least have the choice between warning spam or "siggraph interactive" frame rates :thinking:

AnkeAnke, 2023-02-16:
Actually, scratch that, the resize warning appears on lavapipe too :disappointed:

AnkeAnke, 2023-09-13:
On the new version (0.8.2), I don't see the warning anymore. That is, at least on nvidia, intel does not render anymore.
But I have to admit I did not check beforehand with the previous version, so it might well be a function of changes to my default GPU or my system in general, and not of rerun itself.

So I'll go ahead and close the issue.

AnkeAnke, 2023-09-13:
In case you're wondering: my Intel UHD Graphics 630 gives up with
```
 [time ERROR winit::platform_impl::platform] X11 error: XError {
        description: "BadMatch (invalid parameter attributes)",
        error_code: 8,
        request_code: 149,
        minor_code: 4,
    }
```

Wumpf, 2023-09-13:
this means we don't actually know if this issue is closed, it's *very likely* we still have this problem
@AnkeAnke is it no longer possible to run with nvidia for you?

Wumpf, 2023-09-13:
apologies, didn't see the first message actually. Wgpu has some issues with Intel currently, there's more fixes about that in the next version as far as I know (nvidia optimus drivers are per usually messy)

closed 2023-09-13 by Wumpf