rerun-io/rerun #1320: Cannot install using pip

m-decoster, 2023-02-15:
**Describe the bug**

Despite seemingly meeting the requirements to install the Python version of rerun, I cannot install it using Pip.

The documentation says I need a working version of Python 3.8+. The PyPI website also says >= 3.8. However, this happens when I try to install rerun:

```
$ python3 --version
Python 3.8.5
$ pip --version
pip 23.0 from /home/mcdcoste/envs/rerun_env/lib/python3.8/site-packages/pip (python 3.8)
$ pip install rerun-sdk
ERROR: Could not find a version that satisfies the requirement rerun-sdk (from versions: none)
ERROR: No matching distribution found for rerun-sdk
```

**To Reproduce**
Steps to reproduce the behavior:
1. Get Ubuntu 20.04
2. Install Python 3.8.5
3. Install Pip 23
4. Install rerun-sdk

**Expected behavior**

Expected behavior is that `pip install rerun-sdk` would install the rerun sdk.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

Wumpf, 2023-02-15:
Thank you for reporting! We're looking into this right away!

jleibs, 2023-02-15:
@m-decoster I reproduced this locally in a 20.04 docker container and can confirm it's an issue with how we're building the wheels. Working on an updated builder now that should do the trick for you. Will report back when I have an updated package to try out.

jleibs, 2023-02-15:
Ok, @m-decoster I managed to create a one-off build from an ubuntu 20.04 environment (specifically with glibc-2.31 ABI).  It will take a bit more to get this fully integrated into our CI, but I'm curious if this works for you.

[rerun_glibc_2_31.zip](https://github.com/rerun-io/rerun/files/10746610/rerun_glibc_2_31.zip)

Download the attached zip, then:
```
unzip rerun_glibc_2_31.zip
pip install rerun_sdk-0.2.0-cp38-abi3-manylinux_2_31_x86_64.whl
python -m rerun_demo
```

m-decoster, 2023-02-16:
Thanks @jleibs, this works perfectly!

jleibs, 2023-02-16:
Awesome, thanks for testing, @m-decoster 

I'll go ahead and figure out how to get these changes into our CI then.

robertbagge, 2023-02-17:
@jleibs, is `rerun_glibc_2_31.zip` with the web feature? I get 

```
RuntimeError: The Rerun SDK was not compiled with the 'web' feature
```

when trying to run the objectron demo with `--serve`

jleibs, 2023-02-17:
@robertbagge That test build wasn't, but I just landed the change to do this in our CI.  Looks like I've got one more bug to fix, but should have updated pre-release wheels for you very soon.

jleibs, 2023-02-17:
@robertbagge ok, the CI build finally finished. It won't be available via pypi until our next patch-release, but you can grab the wheel from here: https://github.com/rerun-io/rerun/releases/download/latest/rerun_sdk-0.2.0+df920dc.1-cp38-abi3-manylinux_2_31_x86_64.whl

mont1004, 2023-02-23:
Hi @jleibs , I'm still using Ubuntu 18.04 (glibc 2.27) and the same issue happens. Could you also add the CI pipeline to generate whl for glibc 2.27? I'm looking forward to working with your sdk.

emilk, 2023-02-23:
We haven't officially decided what our minimum supported Ubuntu is (see https://github.com/rerun-io/rerun/issues/1346), but Ubuntu 18.04 was released in 2018 and will loose standard support [on April 2023](https://wiki.ubuntu.com/Releases), so I'm leaning towards that not making the cut.

Have you tried building from source? https://github.com/rerun-io/rerun/tree/main/rerun_py#from-source

mont1004, 2023-02-23:
Fair enough. I'll give it a shot. Thank you.

jleibs, 2023-02-23:
18.04 also does not ship with python 3.8. Which is currently our minimum required python version, so you'll also need to use a non-system python version.

robertbagge, 2023-02-23:
@jleibs, do you have any timeline on when the next patch release will be?

mont1004, 2023-02-23:
I just tried that, however, I got a couple of problems. The biggest thing is that gtk-sys crate requires `gtk+-3.0 >= 3.24` while it is available up to 3.22 for Ubuntu 18.04. There might be some way to install it forcibly, but that idea sounds no good considering the risk of breaking my environment.

I'll consider migration to the latest LTS version (I'd love to do that personally, but it's not an easy choice in the company😕).

jleibs, 2023-02-23:
> @jleibs, do you have any timeline on when the next patch release will be?

@robertbagge There are a few more things we're hoping to land first. Currently looking like sometime mid- next week.

jleibs, 2023-02-23:
> The biggest thing is that gtk-sys crate requires gtk+-3.0 >= 3.24

@mont1004 yes, that seems likely to cause some problems. I was hoping there might be some way for you to build without the gtk dep, i.e. only using the web-viewer, but this is not currently possible (I just created an issue for this for the future though: https://github.com/rerun-io/rerun/issues/1384).

For now I think you're going to be stuck finding a way to work with 20.04 (or 22.04 which is also LTS).

robertbagge, 2023-02-23:
> > @jleibs, do you have any timeline on when the next patch release will be?
> 
> @robertbagge There are a few more things we're hoping to land first. Currently looking like sometime mid- next week.

Cool, that sounds good

emilk, 2023-02-23:
I've creates an issue to track the next release: https://github.com/rerun-io/rerun/issues/1386

cortwave, 2023-03-08:
@emilk seems like new release 0.3.0 didn't fix issue

jleibs, 2023-03-08:
As of rerun-sdk-0.3.0 this is now working. Just confirmed in a new Ubuntu-20.04 docker container.

closed 2023-03-08 by jleibs

kvnptl, 2023-04-10:
> Ok, @m-decoster I managed to create a one-off build from an ubuntu 20.04 environment (specifically with glibc-2.31 ABI). It will take a bit more to get this fully integrated into our CI, but I'm [...]
After running this, I'm getting below error in Ubuntu 20.04
`ERROR: rerun_sdk-0.2.0-cp38-abi3-manylinux_2_31_x86_64.whl is not a supported wheel on this platform.`

kvnptl, 2023-04-10:
> After running this, I'm getting below error in Ubuntu 20.04 `ERROR: rerun_sdk-0.2.0-cp38-abi3-manylinux_2_31_x86_64.whl is not a supported wheel on this platform.`

ok, it works after upgrading pip using `python -m pip install --upgrade pip`

fabienbaradel, 2023-06-02:
Hi,

I am facing the same issue:
```
$ pip3 install rerun-sdk
Collecting rerun-sdk
  Cache entry deserialization failed, entry ignored
  Could not find a version that satisfies the requirement rerun-sdk (from versions: )
No matching distribution found for rerun-sdk
```

Here are some info about my system:
```
$ pip -V
pip 19.2.2 from /nfs/tools/humans/conda/lib/python3.7/site-packages/pip (python 3.7)

$ uname -a
Linux <name_of_my_server> 3.10.0-1160.80.1.el7.x86_64 #1 SMP Tue Nov 8 15:48:59 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

Do you know where the issue come from?
Thanks,

fabienbaradel, 2023-06-05:
Just updated my pip and python to newer version but I am still facing same issue:
```
$ pip -V
pip 23.1.2 from /home/fbaradel/.local/lib/python3.9/site-packages/pip (python 3.9)

$ pip install rerun-sdk
ERROR: Could not find a version that satisfies the requirement rerun-sdk (from versions: none)
ERROR: No matching distribution found for rerun-sdk
```
The output is not verbose so it is difficult to know where the issue comes from.
Any idea @jleibs ?

jleibs, 2023-06-05:
Interesting. What linux version is this? `3.10.0-1160.80.1.el7.x86_64` looks like maybe a Redhat/Fedora variant?

Our packages are built as manylinux_2_31, which works back to ubuntu 20.04. Looks like on the Fedora side this requires Fedora 32 or later.

fabienbaradel, 2023-06-07:
The linux version is Centos. What do you recommend me for installing rerun? Thanks.

nchodosh, 2023-07-05:
I am facing the same issue as @fabienbaradel on CentOS 7:
```
$uname -a
Linux trinity-0-17.eth 3.10.0-957.1.3.el7.x86_64 #1 SMP Thu Nov 29 14:49:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

I'm guessing it's related to OS's ancient glibc version (2.17).

ani0075saha, 2024-03-27:
@jleibs 

I am trying to install rerun sdk on a RHEL system, but it fails. Any suggestions? 

```
pip install rerun-sdk
ERROR: Could not find a version that satisfies the requirement rerun-sdk (from versions: none)
ERROR: No matching distribution found for rerun-sdk

pip install https://files.pythonhosted.org/packages/6b/f6/c79189e4427c19c585da61b9394f30234ac1f70859c61e36cf5f11bb4fa4/rerun_sdk-0.14.1-cp38-abi3-manylinux_2_31_x86_64.whl
ERROR: rerun_sdk-0.14.1-cp38-abi3-manylinux_2_31_x86_64.whl is not a supported wheel on this platform.

cat /etc/os-release
NAME="Red Hat Enterprise Linux"
VERSION="8.9 (Ootpa)"
ID="rhel"
ID_LIKE="fedora"
VERSION_ID="8.9"
PLATFORM_ID="platform:el8"
PRETTY_NAME="Red Hat Enterprise Linux 8.9 (Ootpa)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:redhat:enterprise_linux:8::baseos"
HOME_URL="https://www.redhat.com/"
DOCUMENTATION_URL="https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/8"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 8"
REDHAT_BUGZILLA_PRODUCT_VERSION=8.9
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="8.9"

uname -a
Linux <xxx> 4.18.0-513.18.1.el8_9.x86_64 #1 SMP Thu Feb 1 03:51:05 EST 2024 x86_64 x86_64 x86_64 GNU/Linux
```

jleibs, 2024-03-27:
@ani0075saha our current pypi-hosted wheel depends on glibc 2.31.  It looks like RHEL-8.9 is still using glibc-2.28 

For older OS's such as redhat 8 I would currently recommend either doing a source install, or using using our `conda-forge` package (https://prefix.dev/channels/conda-forge/packages/rerun-sdk)

shubham-monarch, 2024-09-21:
upgrading the pip should fix this 
`pip install --upgrade pip`

feisuo, 2024-11-20:
[image]

I am facing the same issue with jetson 
[image]

jleibs, 2024-11-25:
@feisuo I've confirmed your exact install command works for me on a modern tegra platform from within the lerobot conda environment.
```
$ uname -a
Linux ubuntu 5.10.192-tegra #1 SMP PREEMPT Mon Feb 19 20:19:53 PST 2024 aarch64 aarch64 aarch64 GNU/Linux
```

My guess is that 4.9.337-tegra has too old of a glibc.  Our pypi-packaged wheels require glibc >= 2.31.  (You can check this by running `ldd --version`)

I believe our conda-forge packages may work with an older glibc. You might try `conda install rerun-sdk` instead of using the pip packages.

arjunskumar, 2024-12-30:
this should fix 
```
pip install pipx
pipx install rerun-sdk
```

narutojxl, 2025-03-14:
Hello, @arjunskumar
When after installed rerun with pipx, when using rerun in python script with  `import rerun as rr`, it outputs  `ModuleNotFoundError: No module named 'rerun'`, how to fix this lib error, thanks?
 I found rerun installed in `/home/eisonjiao/.pixi/envs/rerun-sdk`, also added `export PATH="/home/eisonjiao/.pixi/bin:$PATH"` and  `export PYTHONPATH="$PYTHONPATH:/home/eisonjiao/.pixi/envs/rerun-sdk/"` into `~/.bashrc`.

narutojxl, 2025-03-14:
When i changed python into 3.8, installed rerun with pipx, add  `export PATH="/home/eisonjiao/.local/share/pipx/venvs/rerun-sdk/bin/rerun:$PATH"` and  `export PYTHONPATH="$PYTHONPATH:/home/eisonjiao/.local/share/pipx/venvs/rerun-sdk/lib/python3.8/site-packages"` into `~/.bashrc`. i can run `rerun` in terminal, but also could not find `ModuleNotFoundError: No module named 'rerun'` when running python script with `import rerun as rr`.

>

    eisonjiao@eisonjiao-Z3-Air-Series-GM5TGEO:~/show_project$ pip install rerun-sdk
    Requirement already satisfied: rerun-sdk in /home/eisonjiao/.local/share/pipx/venvs/rerun-sdk/lib/python3.8/site-packages (0.22.1)
    Requirement already satisfied: pyarrow>=14.0.2 in /home/eisonjiao/.local/share/pipx/venvs/rerun-sdk/lib/python3.8/site-packages (from rerun-sdk) (17.0.0)
    Requirement already satisfied: typing-extensions>=4.5 in /home/eisonjiao/.local/share/pipx/venvs/rerun-sdk/lib/python3.8/site-packages (from rerun-sdk) (4.12.2)
    Requirement already satisfied: numpy>=1.23 in /home/eisonjiao/.local/share/pipx/venvs/rerun-sdk/lib/python3.8/site-packages (from rerun-sdk) (1.24.4)
    Requirement already satisfied: pillow>=8.0.0 in /home/eisonjiao/.local/share/pipx/venvs/rerun-sdk/lib/python3.8/site-packages (from rerun-sdk) (10.4.0)
    Requirement already satisfied: attrs>=23.1.0 in /home/eisonjiao/.local/share/pipx/venvs/rerun-sdk/lib/python3.8/site-packages (from rerun-sdk) (25.3.0)
    eisonjiao@eisonjiao-Z3-Air-Series-GM5TGEO:~/show_project$