rerun-io/rerun #1323: Mechanism to force image to be treated as a tensor

jleibs, 2023-02-16:
Not all 2D tensors are images. Data like an stft looks like an image but doesn't play nicely with the image viewer.

The tensor viewer has a number of nice features like:
 - support for f64 data
 - the ability to not lock the aspect ratio
 - support for colormaps

We can trick the viewer into creating a tensor by adding an empty outer dimension.

Example:
```
f, t, Zxx = scipy.signal.stft(x, fs, nperseg=1000)
Zxx_image = np.abs(Zxx).astype(np.float32)
Zxx_tensor = np.expand_dims(np.abs(Zxx), axis=0)

rr.init("stft_test", spawn=True)
rr.log_tensor("stft_image", Zxx_image)
rr.log_tensor("stft_tensor", Zxx_tensor)
```
[image]

By comparison, using the image viewer we end up with:

Wumpf, 2023-02-16:
Related to https://github.com/rerun-io/rerun/issues/1164

jleibs, 2023-11-28:
```
rr.log("stft_tensor", rr.Tensor(Zxx_image))
```
now does exactly what this describes

closed 2023-11-28 by jleibs