rerun-io/rerun #1324: Crash when running across SSH with X11 forwarding

mchhoy, 2023-02-16:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Enable X11 forwarding on the server:

```
nano /etc/ssh/sshd_config.d/ssh_user.conf
```

Use the following sshd_config, but use a new port:

```
X11Forwarding yes
```

Then

```
sudo service ssh restart
```
Then on the client

ssh [address] -X

2. `cargo install rerun`
3. `WINIT_UNIX_BACKEND=x11 rerun`

(I tried this idea https://github.com/rust-windowing/winit/issues/305)

```
$ WINIT_UNIX_BACKEND=x11 rerun
2023-02-16T07:42:09.613480Z  INFO re_sdk_comms::server: Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
2023-02-16T07:42:09.657696Z  INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 1    
Segmentation fault (core dumped)
```

**Expected behavior**
X11 forwarding should show the screen on the client

**Desktop (please complete the following information):**
 - OS: 18.04.1-Ubuntu (server and client)

mchhoy, 2023-02-16:
With `RUST_LOG=trace`:

```
RUST_LOG=trace WINIT_UNIX_BACKEND=x11 rerun
2023-02-16T08:01:42.464579Z  INFO re_sdk_comms::server: Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
2023-02-16T08:01:42.464659Z DEBUG eframe: Using the wgpu renderer
2023-02-16T08:01:42.506455Z TRACE mio::poll: registering event source with poller: token=Token(0), interests=READABLE    
2023-02-16T08:01:42.507497Z DEBUG eframe::native::run: Entering the winit event loop (run_return)…
2023-02-16T08:01:42.507529Z DEBUG eframe::native::file_storage: Loading app state from "/home/michael/.local/share/rerunviewer/app.ron"…
2023-02-16T08:01:42.510815Z  INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 1    
2023-02-16T08:01:42.510825Z DEBUG winit::platform_impl::platform::x11::window: Calculated physical dimensions: 1600x1200    
Segmentation fault (core dumped)
``

mchhoy, 2023-02-16:
Also note I currently use

https://github.com/sebcrozet/kiss3d

with the same setup and it works fine

Wumpf, 2023-02-16:
@mchhoy thank you for logging this issue! So far fairly in the dark what could cause this, haven't seen this crash so far.
I just landed a PR (#1332) that allows us to force a different graphics backend. If you don't mind and have the time, could you please run 
`WGPU_BACKEND=gl RUST_LOG=wgpu_hal=debug,wgpu=debug cargo run`
from latest main and report back what it logs out?

Thank you!

mchhoy, 2023-02-17:
Erm some other error now .. also I forgot to note that I am in a docker container.

```
/home/michael/rerun/crates/rerun$ cargo --version
cargo 1.67.1 (8ecd4f20a 2023-01-10)

/home/michael/rerun/crates/rerun$ WGPU_BACKEND=gl RUST_LOG=wgpu_hal=debug,wgpu=debug cargo run
    Finished dev [optimized + debuginfo] target(s) in 0.28s
     Running `/host/mnt/data_1/michael/rerun/target/debug/rerun`
2023-02-17T02:03:52.492017Z DEBUG wgpu_hal::gles::egl: Client extensions: [
    "EGL_EXT_device_base",
    "EGL_EXT_device_enumeration",
    "EGL_EXT_device_query",
    "EGL_EXT_platform_base",
    "EGL_KHR_client_get_all_proc_addresses",
    "EGL_EXT_client_extensions",
    "EGL_KHR_debug",
    "EGL_EXT_platform_device",
    "EGL_EXT_platform_wayland",
    "EGL_KHR_platform_wayland",
    "EGL_EXT_platform_x11",
    "EGL_KHR_platform_x11",
    "EGL_MESA_platform_xcb",
    "EGL_MESA_platform_gbm",
    "EGL_KHR_platform_gbm",
    "EGL_MESA_platform_surfaceless",
]    
2023-02-17T02:03:52.492090Z  INFO wgpu_hal::gles::egl: Loading Wayland library to get the current display    
error: XDG_RUNTIME_DIR not set in the environment.
2023-02-17T02:03:52.492144Z  INFO wgpu_hal::gles::egl: Loading X11 library to get the current display    
2023-02-17T02:03:52.494892Z  INFO wgpu_hal::gles::egl: Using X11 platform    
2023-02-17T02:03:52.494930Z  INFO wgpu_hal::gles::egl: Enabling EGL debug output    
libEGL warning: DRI3: failed to query the version
libEGL warning: DRI2: failed to authenticate
2023-02-17T02:03:52.538479Z  INFO wgpu_hal::gles::egl: Display vendor "Mesa Project", version (1, 5)    
2023-02-17T02:03:52.538514Z DEBUG wgpu_hal::gles::egl: Display extensions: [
    "EGL_EXT_create_context_robustness",
    "EGL_KHR_cl_event2",
    "EGL_KHR_config_attribs",
    "EGL_KHR_context_flush_control",
    "EGL_KHR_create_context",
    "EGL_KHR_create_context_no_error",
    "EGL_KHR_fence_sync",
    "EGL_KHR_get_all_proc_addresses",
    "EGL_KHR_gl_colorspace",
    "EGL_KHR_gl_renderbuffer_image",
    "EGL_KHR_gl_texture_2D_image",
    "EGL_KHR_gl_texture_3D_image",
    "EGL_KHR_gl_texture_cubemap_image",
    "EGL_KHR_image_base",
    "EGL_KHR_no_config_context",
    "EGL_KHR_reusable_sync",
    "EGL_KHR_surfaceless_context",
    "EGL_EXT_pixel_format_float",
    "EGL_KHR_wait_sync",
    "EGL_MESA_configless_context",
    "EGL_MESA_drm_image",
    "EGL_MESA_query_driver",
]    
2023-02-17T02:03:52.538551Z  INFO wgpu_hal::gles::egl:  EGL surface: +srgb    
2023-02-17T02:03:52.538558Z  INFO wgpu_hal::gles::egl:  Trying native-render    
2023-02-17T02:03:52.538590Z  INFO wgpu_hal::gles::egl:  EGL context: +debug    
2023-02-17T02:03:52.538599Z  INFO wgpu_hal::gles::egl:  EGL context: +robust access    
2023-02-17T02:03:52.545500Z  INFO wgpu_hal::gles::egl:  EGL context: +surfaceless    
2023-02-17T02:03:52.548219Z  INFO wgpu_hal::gles::egl: Max label length: 256    
2023-02-17T02:03:52.548231Z  INFO wgpu_hal::gles::egl: Enabling GLES debug output    
2023-02-17T02:03:52.548255Z  INFO wgpu_hal::gles::adapter: Vendor: Mesa/X.org    
2023-02-17T02:03:52.548260Z  INFO wgpu_hal::gles::adapter: Renderer: llvmpipe (LLVM 12.0.0, 256 bits)    
2023-02-17T02:03:52.548264Z  INFO wgpu_hal::gles::adapter: Version: OpenGL ES 3.2 Mesa 21.2.6    
2023-02-17T02:03:52.548269Z DEBUG wgpu_hal::gles::adapter: Extensions: {
[... 328 lines cut ...]
             at ./src/run.rs:235:9
  31: rerun::run::run::{{closure}}
             at ./src/run.rs:133:36
  32: rerun::main::{{closure}}
             at ./src/main.rs:11:9
  33: tokio::runtime::park::CachedParkThread::block_on::{{closure}}
             at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.24.1/src/runtime/park.rs:283:63
  34: tokio::runtime::coop::with_budget
             at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.24.1/src/runtime/coop.rs:102:5
  35: tokio::runtime::coop::budget
             at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.24.1/src/runtime/coop.rs:68:5
  36: tokio::runtime::park::CachedParkThread::block_on
             at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.24.1/src/runtime/park.rs:283:31
  37: tokio::runtime::context::BlockingRegionGuard::block_on
             at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.24.1/src/runtime/context.rs:315:13
  38: tokio::runtime::scheduler::multi_thread::MultiThread::block_on
             at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.24.1/src/runtime/scheduler/multi_thread/mod.rs:66:9
  39: tokio::runtime::runtime::Runtime::block_on
             at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.24.1/src/runtime/runtime.rs:284:45
  40: rerun::main
             at ./src/main.rs:10:5
  41: core::ops::function::FnOnce::call_once
             at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/ops/function.rs:507:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

teh-cmc, 2023-02-17:
You're hitting #1258 @mchhoy (and #1322).

I'll fix that today, in the meantime you can `cd` to the root of the workspace, `cargo run` once again, and you should be good to go.

teh-cmc, 2023-02-17:
:point_up: Both issues should be fixed on latest `main`.

mchhoy, 2023-02-20:
Last time I couldn't build from the workspace root as the web viewer crate didn't compile, maybe because there is no browser available? 

```
 WGPU_BACKEND=gl RUST_LOG=wgpu_hal=debug,wgpu=debug cargo run --release

OR

WGPU_BACKEND=gl RUST_LOG=wgpu_hal=debug,wgpu=debug cargo run
```

[image]

But without WGPU_BACKEND=gl

```
RUST_LOG=wgpu_hal=debug,wgpu=debug cargo run
    Finished dev [optimized + debuginfo] target(s) in 0.29s
     Running `/host/mnt/data_1/michael/rerun/target/debug/rerun`
2023-02-20T02:26:44.032310Z  INFO wgpu_hal::vulkan::instance: Unable to find extension: VK_EXT_swapchain_colorspace    
2023-02-20T02:26:44.032865Z  WARN wgpu_hal::vulkan::instance: Unable to find layer: VK_LAYER_KHRONOS_validation    
2023-02-20T02:26:44.040622Z  INFO wgpu_hal::vulkan::instance: Instance version: 0x402083    
2023-02-20T02:26:44.040636Z  INFO wgpu_hal::vulkan::instance: Enabling debug utils    
2023-02-20T02:26:44.040660Z  INFO wgpu_hal::vulkan::instance: Enabling device properties2    
2023-02-20T02:26:44.042522Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        loader_add_implicit_layer: Disabling implicit layer VK_LAYER_MESA_device_select for using an old API version 1.2 versus application requested 1.3    
2023-02-20T02:26:44.042539Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042573Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        loader_add_implicit_layer: Disabling implicit layer VK_LAYER_MESA_device_select for using an old API version 1.2 versus application requested 1.3    
2023-02-20T02:26:44.042579Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042587Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_8bit_storage (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.1    
2023-02-20T02:26:44.042593Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042599Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_16bit_storage (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.1    
2023-02-20T02:26:44.042605Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042611Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_bind_memory2 (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.1    
2023-02-20T02:26:44.042617Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042623Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_buffer_device_address (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.1    
2023-02-20T02:26:44.042628Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042634Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_copy_commands2 (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.1    
2023-02-20T02:26:44.042640Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042646Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_create_renderpass2 (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.1    
2023-02-20T02:26:44.042651Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042657Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_dedicated_allocation (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.3    
2023-02-20T02:26:44.042662Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042668Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_descriptor_update_template (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.1    
2023-02-20T02:26:44.042674Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042680Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_device_group (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.4    
2023-02-20T02:26:44.042685Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042691Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_draw_indirect_count (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.1    
2023-02-20T02:26:44.042697Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
2023-02-20T02:26:44.042704Z  INFO wgpu_hal::vulkan::instance: GENERAL [Loader Message (0x0)]
        Device Extension: VK_KHR_driver_properties (/usr/lib/x86_64-linux-gnu/libvulkan_lvp.so) version 0.0.1    
2023-02-20T02:26:44.042711Z  INFO wgpu_hal::vulkan::instance:   objects: (type: INSTANCE, hndl: 0x560d8a8a8620, name: ?)    
[... cut ...]
  22: tokio::runtime::runtime::Runtime::block_on
             at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.24.1/src/runtime/runtime.rs:284:45
  23: rerun::main
             at src/main.rs:10:5
  24: core::ops::function::FnOnce::call_once
             at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/ops/function.rs:507:5
      
Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting

Segmentation fault (core dumped)
```

mchhoy, 2023-02-20:
BTW the error when building the workspace root instead of `crates/rerun` is:

```
 Build web viewer wasm…
  wasm build cmd: "cargo" "build" "--target-dir" "/host/mnt/data_1/michael/rerun/target_wasm" "-p" "re_viewer" "--lib" "--target" "wasm32-unknown-unknown"
     Compiling re_renderer v0.2.0 (/host/mnt/data_1/michael/rerun/crates/re_renderer)
     Compiling re_viewer v0.2.0 (/host/mnt/data_1/michael/rerun/crates/re_viewer)
      Finished dev [optimized + debuginfo] target(s) in 14.54s
  compile status: exit status: 0
  compile stderr: 
  wasm-bindgen cmd: "wasm-bindgen" "/host/mnt/data_1/michael/rerun/target_wasm/wasm32-unknown-unknown/debug/re_viewer.wasm" "--out-dir" "/host/mnt/data_1/michael/rerun/crates/re_web_server/../../web_viewer" "--no-modules" "--no-typescript"
  thread 'main' panicked at 'Failed to generate JS bindings: No such file or directory (os error 2). target_path: "/host/mnt/data_1/michael/rerun/target_wasm/wasm32-unknown-unknown/debug/re_viewer.wasm", build_dir: /host/mnt/data_1/michael/rerun/crates/re_web_server/../../web_viewer', crates/re_web_server/build.rs:193:31
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/std/src/panicking.rs:575:5
     1: core::panicking::panic_fmt
               at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/panicking.rs:64:14
     2: build_script_build::build_web::{{closure}}
               at ./build.rs:193:31
     3: core::result::Result<T,E>::unwrap_or_else
               at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/result.rs:1504:23
     4: build_script_build::build_web
               at ./build.rs:189:18
     5: build_script_build::main
               at ./build.rs:269:9
     6: core::ops::function::FnOnce::call_once
               at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/ops/function.rs:507:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
warning: build failed, waiting for other jobs to finish...
```

mchhoy, 2023-02-20:
BTW this is the GDB backtrace

```
Thread 1 "rerun" received signal SIGSEGV, Segmentation fault.
0x00007fffbcbc8734 in ?? () from /usr/lib/x86_64-linux-gnu/libvulkan_lvp.so
(gdb) 
(gdb) 
(gdb) bt
#0  0x00007fffbcbc8734 in  () at /usr/lib/x86_64-linux-gnu/libvulkan_lvp.so
#1  0x00007fffbcbc9015 in  () at /usr/lib/x86_64-linux-gnu/libvulkan_lvp.so
#2  0x000055555630066d in ash::extensions::khr::surface::Surface::get_physical_device_surface_support (physical_device=..., queue_family_index=0, surface=..., self=<optimized out>)
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/ash-0.37.2+1.3.238/src/extensions/khr/surface.rs:32
#3  wgpu_hal::vulkan::adapter::<impl wgpu_hal::Adapter<wgpu_hal::vulkan::Api> for wgpu_hal::vulkan::Adapter>::surface_capabilities (self=<optimized out>, surface=0x37dec140a00) at src/vulkan/adapter.rs:1540
#4  0x00005555561c6c14 in wgpu_core::instance::<impl wgpu_core::hub::Global<G>>::request_adapter::gather::{{closure}} (exposed=<optimized out>)
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-core-0.15.0/src/instance.rs:743
#5  alloc::vec::Vec<T,A>::retain::{{closure}} (elem=<optimized out>) at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/alloc/src/vec/mod.rs:1562
#6  alloc::vec::Vec<T,A>::retain_mut::process_loop (original_len=1, g=0x7fffffff5d00, f=<optimized out>) at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/alloc/src/vec/mod.rs:1642
#7  alloc::vec::Vec<T,A>::retain_mut (self=0x7fffffff5e08, f=...) at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/alloc/src/vec/mod.rs:1668
#8  alloc::vec::Vec<T,A>::retain (self=0x7fffffff5e08, f=...) at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/alloc/src/vec/mod.rs:1562
#9  0x000055555620ddc7 in wgpu_core::instance::<impl wgpu_core::hub::Global<G>>::request_adapter::gather
    (instance=..., inputs=<optimized out>, compatible_surface=..., force_software=false, device_types=0x7fffffff5e98)
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-core-0.15.0/src/instance.rs:739
#10 0x000055555628bb39 in wgpu_core::instance::<impl wgpu_core::hub::Global<G>>::request_adapter (self=0x37dec375010, desc=0x7fffffff7520, inputs=...)
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-core-0.15.0/src/instance.rs:769
#11 0x00005555562c1a62 in <wgpu::backend::direct::Context as wgpu::context::Context>::instance_request_adapter (options=<optimized out>, self=<optimized out>) at src/backend/direct.rs:576
#12 <T as wgpu::context::DynContext>::instance_request_adapter (self=0x5555575e0ab8, options=<optimized out>) at src/context.rs:1932
#13 0x000055555620ea98 in wgpu::Instance::request_adapter (self=<optimized out>, options=0x0) at src/lib.rs:1424
#14 0x0000555555eccff1 in egui_wgpu::winit::Painter::ensure_render_state_for_surface::{{closure}} () at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/egui-wgpu-0.21.0/src/winit.rs:107
#15 egui_wgpu::winit::Painter::set_window::{{closure}} () at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/egui-wgpu-0.21.0/src/winit.rs:185
#16 pollster::block_on (fut=...) at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/pollster-0.3.0/src/lib.rs:128
#17 0x0000555555e9fcb4 in eframe::native::run::wgpu_integration::WgpuWinitApp::init_run_state (self=0x7fffffff9600, event_loop=0x37dec1b2c10, storage=..., window=...) at src/native/run.rs:1114
#18 <eframe::native::run::wgpu_integration::WgpuWinitApp as eframe::native::run::WinitApp>::on_event (self=0x7fffffff9600, event_loop=0x37dec1b2c10, event=<optimized out>) at src/native/run.rs:1302
#19 0x0000555555ed96cc in eframe::native::run::run_and_return::{{closure}} (event=..., event_loop=0x0, control_flow=0x7fffffff8b60) at src/native/run.rs:161
#20 0x0000555555e7c131 in winit::platform_impl::platform::sticky_exit_callback (evt=..., target=0x37dec1b2c10, control_flow=0x7fffffff8b60, callback=0x7fffffff8da8)
--Type <RET> for more, q to quit, c to continue without paging--
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.28.1/src/platform_impl/linux/mod.rs:884
#21 winit::platform_impl::platform::x11::EventLoop<T>::run_return::single_iteration (this=0x7ffff5f9daf0, control_flow=0x7fffffff8b60, cause=0x7fffffff8c80, callback=0x7fffffff8da8)
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.28.1/src/platform_impl/linux/x11/mod.rs:334
#22 0x0000555555e7c851 in winit::platform_impl::platform::x11::EventLoop<T>::run_return (self=0x7ffff5f9daf0, callback=...)
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.28.1/src/platform_impl/linux/x11/mod.rs:443
#23 0x0000555555e75282 in winit::platform_impl::platform::EventLoop<T>::run_return (self=0x7ffff5f9daf0, callback=...)
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.28.1/src/platform_impl/linux/mod.rs:785
#24 <winit::event_loop::EventLoop<T> as winit::platform::run_return::EventLoopExtRunReturn>::run_return (self=0x7ffff5f9daf0, event_handler=...)
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.28.1/src/platform/run_return.rs:51
#25 eframe::native::run::run_and_return (event_loop=0x7ffff5f9daf0, winit_app=...) at src/native/run.rs:124
#26 0x0000555555ea729c in eframe::native::run::wgpu_integration::run_wgpu::{{closure}} (event_loop=0x7ffff5f9daf0, native_options=...) at src/native/run.rs:1410
[... cut ...]
#42 tokio::runtime::scheduler::multi_thread::MultiThread::block_on (self=<optimized out>, handle=<optimized out>, future=...)
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.24.1/src/runtime/scheduler/multi_thread/mod.rs:66
#43 0x00005555557ad8f0 in tokio::runtime::runtime::Runtime::block_on (self=<optimized out>, future=...)
    at /home/michael/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.24.1/src/runtime/runtime.rs:284
#44 0x00005555557ad634 in rerun::main () at crates/rerun/src/main.rs:10
```

mchhoy, 2023-02-20:
This is for 96437ee8979a2b629526875b515684bebb2e69c6