rerun-io/rerun #1325: Threading support

emilk, 2023-02-16:
The Rerun viewer needs to run in Wasm on the browser. This limits our threading ability. `std::thread` is, for instance, not available. At the same time we want to make sure our viewer achieves good frame rate when compiled natively.

## Parallelism on native, with single-threaded fallback on web
`rayon` is the go-to Rust crate for fine-grained parallelism. It is commonly used by library crates (behind a feature flag).
Currently `rayon` does not run in wasm in the browser, but this might change soon:

* https://github.com/rayon-rs/rayon/issues/861
* https://github.com/rayon-rs/rayon/pull/1019

In the mean-time we can also define our own helpers for parallelism, e.g.:

``` rust
/// Run two things in parallel
#[cfg(not(target_arch = "wasm32"))]
fn parallel<A: Send, B: Send>(
    a: impl Send + FnOnce() -> A,
    b: impl Send + FnOnce() -> B,
) -> (A, B) {
    rayon::join(a, b)
}

// Serial-fallback for wasm32
#[cfg(target_arch = "wasm32")]
fn parallel2<A: Send, B: Send>(
    a: impl Send + FnOnce() -> A,
    b: impl Send + FnOnce() -> B,
) -> (A, B) {
    ((a)(), (b)()) 
}
```

## Parallelism on wasm32
With a bit of work, one can spawn wasm32 threads in the browser with shared memory:

https://www.tweag.io/blog/2022-11-24-wasm-threads-and-messages/

Unfortunately, this requires a nightly compiler, afaict. I think it could be fine for use to rely on the nightly compiler for just wasm32.

I suggest we keep this use-case in mind, but focus on the native performance for now.

emilk, 2023-02-28:
Thanks @RReverser - I forgot to mention that! We experimented with [wasm-bindgen-rayon](https://github.com/GoogleChromeLabs/wasm-bindgen-rayon) last autumn but decided we didn't want to commit to nightly back then.

RReverser, 2023-02-28:
Yeah fair enough, I wish rebuilding standard library wasn't necessary. There is one trick you can use to build even on stable, although it's not particularly recommended: https://twitter.com/RReverser/status/1608840842654908419

emilk, 2023-11-06:
We have started using `rayon` in a few places in the viewer, relying on its single-threaded fallback on web.

In order to get a proper speedup we need to make our `RenderContext` sync

PR #4460 (2023-12-11): Parallize Space View system evaluation
### What

* Fixes #1325 

Two levels of parallism actually:
* All context & part systems for a single space view each run in parallel
* All space views now in parallel first setup their query and then kick of their (parallized!) system execution

This gives us major speedups whenever:
* There is several heavy systems in a scene (e.g. a lot of points and a lot of arrows)
* There is several space views

As such this is a very fundamental & widely applicable parallelization!

[image]
(a rather silly "points with arrows" scene duplicated a bunch of times. Not [... cut]
commit ea4dff1e (2023-12-11): Parallize Space View system evaluation (#4460)

### What

* Fixes #1325 

Two levels of parallism actually:
* All context & part systems for a single space view each run in
parallel
* All space views now in parallel first setup their query and then kick
of their (parallized!) system execution

This gives us major speedups whenever:
* There is several heavy systems in a scene (e.g. a lot of points and a
lot of arrows)
* There is several space views

As such this is a very fundamental & widely applicable parallelization!

[image]
(a rather silly "points with arrows" scene duplicated a bunch of t [... cut]
closed 2023-12-11 by Wumpf