rerun-io/rerun #1326: C and C++ API:s

emilk, 2023-02-16:
Let's start with a low-level C API, and then build higher level API:s on top of it, including C++ API.

## Low-level C API
For the lowest level C API we need only some basic functions for:

* set app id and recording id
* toggle loggling on/off
* spawn, connect, serve, spawn
* send arrow message

Arrow messages can be built with the [Arrow C data interface](https://arrow.apache.org/docs/format/CDataInterface.html).

Of course, building an arrow message by hand for e.g. a point-cloud is a lot of work using the C API, so we quickly want to add helpers.

## Mid-level C API
Here we add helper functions for building arrow messages from or primitives, i.e. point clouds, images, etc.

## High-level C API
Here we add integration with e.g. OpenCV to make dumping a C image a one-line call.

## High-level C++ API
This is a wrapper around our C API which makes usage a bit more ergonomic for C++ users.

asuper0, 2023-06-13:
How is this going on? I'd like to write C API which is wrapper of RUST API, and cover all the primitives. Expect your advise.

crownedone, 2023-06-13:
It looks like they are still working on additional primitives (i.e #1630) and mentioned that they think about changing the structure of these in a way that makes interfacing with other languages easier.

That means writing a C interface now could be made obsolete/changed soon (please correct me).

Is there any more specific information on this? (Eagerly waiting to be able to use Rerun from C++!!!)

Wonder if you considered the use of some general data interface (like Protobuf)?

I'd be willing to contribute as well ;)

Wumpf, 2023-06-13:
@asuper0 @crownedone We're making some good progress on generating Rust & Python APIs from data definition files since we figured by now that adding new components _and_ new language front ends is way too tedious otherwise. I sadly still can't give any timelines, but this gets us a _huge_ step towards supporting more languages and a C interface is going to be the first one added.

I deeply appreciate the willingness to help 🙇 , but given that we commited to that way now, anything landing regarding C is pretty much blocked by the codegen effort.
(There's plenty of other places where we could use help though, check the [Good First Issue](https://github.com/rerun-io/rerun/issues?q=is%3Aissue+is%3Aopen+label%3A%22good+first+issue%22) and the [Quick Issue](https://github.com/rerun-io/rerun/issues?q=is%3Aissue+is%3Aopen+label%3A%22%F0%9F%8F%8E%EF%B8%8F+Quick+Issue%22) tags 😃 )

That said, if you want to pipe through something to C the easiest way right now is to write a Rust library depending on the Rerun crate that exposes what you need to C.

If you want to track the progress / are curious what the upcoming codegen things looks like check [PRs](https://github.com/rerun-io/rerun/pulls?q=is%3Apr+label%3Acodegen%2Fidl+) and [Issues](https://github.com/rerun-io/rerun/issues?q=is%3Aissuelabel%3Acodegen%2Fidl+) tagged with `codegen`

asuper0, 2023-06-14:
> (There's plenty of other places where we could use help though, check the [Good First Issue](https://github.com/rerun-io/rerun/issues?q=is%3Aissue+is%3Aopen+label%3A%22good+first+issue%22) and the [...]
These issue may be too complicated for me🤣 I don't intend to go into the inner code.

My planning is writing a Rust proxy, which offers C-like API that receive raw data and convert them to rerun primitive datatype. 

While I had a experience of upgrading my code dependency of Rerun from 0.4.0 -> 0.5.1 -> 0.6.0, it could be promised that the Rust  proxy code will only have littte change with Rerun upgrading, and have no need to change the C API.

I think these work would't be too much.

Wumpf, 2023-08-07:
New tracking issue for first version of C++ SDK that is posed to be available in 0.9.0:
* #2919
Keeping this open since it's the "original" and a bunch of people might monitor this

emilk, 2023-11-06:
We now have a C++ API!

closed 2023-11-06 by emilk