rerun-io/rerun #1329: Enable clearing data

AnkeAnke, 2023-02-16:
**Is your feature request related to a problem? Please describe.**
I often find myself in a situation where I want to permanently clear up the full history, or at least a certain path, from the viewer. Right now, I work around that by re-opening the viewer and re-populating the viewport so that only paths with data remain in the view. 

Especially when setting up my code for logging data to the viewer, I logged a lot of things I didn't intend to keep. Think misspelled paths, 1000000 single lines, some debug data etc. They all appear in the blueprint list when opening the viewer anew.

As another example, when adding more and more data (think `line0`, `line1`,... `line1000` successively), I can nicely see the set build up in the history. But when I run the same logging code again without somehow clearing the data, the lines remain from prior frames.

**Describe the solution you'd like**
Ideally, I'd like to be able to clear all data in a certain blueprint group in the viewer.

**Describe alternatives you've considered**
I had expected there to be a `New` option in the main menu (above `Open...`), that would just set a clean slate. Alternatively, a `Clear all data` button somewhere would do. Or at least a function to call, e.g. `rr.remove_all_data('maybe/a/path')`.

**Additional context**
Example: First stared logging the red lines one by one (which nicely build up when I play the history back), then the blue lines (same code with changed parameters, thus same paths). I have no way to clear the red lines out in-between except by closing and re-opening the viewer. Thus, I get both red and blue lines.
[image]

teh-cmc, 2023-02-16:
Thanks for the thorough report :pinched_fingers: 

Having some control over this directly within the viewer would definitely be nice.

In the meantime, [`rr.log_cleared`](https://ref.rerun.io/docs/python/v0.2.0/package/rerun/__init__/#rerun.log_cleared) should get you almost all the way there.
You can see it in action in the [`api_demo` example](https://github.com/rerun-io/rerun/blob/main/examples/python/api_demo/main.py#L131): `python examples/python/api_demo/main.py --demo log_cleared`.

It only affects the visibility of the data though, not its existence, i.e. it will clear out what you see in the scene but the entity paths themselves will still appear in the blueprint and menus.

marisancans, 2023-06-29:
I have similar problem.
I have 20 minute long video. I take each frame and process it with something, then draw the results with opencv and display it with `rr.log_image("frame", frame_np)`

This slowly fills up my ram and eventually dies. 
I see the same problem in [example](https://github.com/rerun-io/rerun/tree/main/examples/python/live_camera_edge_detection) and the solution there is to just set ram limit?

What I would like to do is:
1. Clear all the frames that are stored somewhere (I assume they are, because I can see thumbnalis in timeline of each frame). I would like to clear them every N frames or do something like a sliding window where I can pause at any time and still see some history. I think rerun has a lot of real time app usages.

2. I would like to clear the whole session state. I leave my rerun app openned and run the python script with each new video I have to process, but I see that the ram is still occupied from previous video. Sometimes when I change the code, say I add new plot or remove it, the session doesnt clear it and I have to reopen the app, which is annoying.

silencht, 2024-11-26:
Any progress on this issue so far, please?
I currently have to use `--memory-limit` to make it work.