rerun-io/rerun #1336: Unable to run on fedora

andrsbtrg, 2023-02-16:
**Describe the annoyance**
First I wanted to say thanks for making open source such an awesome project. So, I am trying to follow the rust quick start, but when trying to build with `cargo run` I get
``warning: `"pkg-config" "--libs" "--cflags" "atk" "atk >= 2.28"` did not exit successfully: exit status: 1``
followed by
 ``error: failed to run custom build command for `atk-sys v0.16.0``
``error: failed to run custom build command for `gdk-sys v0.16.0``
``error: failed to run custom build command for `cairo-sys-rs v0.16.3``
``error: failed to run custom build command for `pango-sys v0.16.3``

**To Reproduce**
Steps to reproduce the behavior:
1. Cargo new-project
2. Cargo add rerun
3. Cargo build
4. Error

**Expected behavior**
Expected successful build

**Your goals**
 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Fedora 37
 -  Kernel: 6.1.9-200.fc37.x86_64
 - Wayland

**Additional context**
I followed the troubleshooting and installed:
```dnf install clang clang-devel clang-tools-extra libxkbcommon-devel pkg-config openssl-devel libxcb-devel```

jleibs, 2023-02-16:
@andrsbtrg it looks like on Fedora the packages are named slightly differently. I was able to get things working by installing the following additional packages in a Fedora 37 docker container: `glib2-devel atk-devel gdk-pixbuf2-devel pango-devel gtk3-devel`

Curious if this works for you.

andrsbtrg, 2023-02-16:
@jleibs thanks for the reply. I was able to build after installing the additional packages. However after running now I get immediatly:
```
cargo run
   Compiling test-rerun v0.1.0 (/home/andres/src/test-rerun)
    Finished dev [unoptimized + debuginfo] target(s) in 24.15s
     Running `target/debug/test-rerun`
thread 'main' panicked at 'Error in Surface::configure: parent device is lost', /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-0.15.1/src/backend/direct.rs:316:9
stack backtrace:
   0: rust_begin_unwind
             at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/std/src/panicking.rs:575:5
   1: core::panicking::panic_fmt
             at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/panicking.rs:64:14
   2: wgpu::backend::direct::Context::handle_error_fatal
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-0.15.1/src/backend/direct.rs:316:9
   3: <wgpu::backend::direct::Context as wgpu::context::Context>::surface_configure
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-0.15.1/src/backend/direct.rs:742:13
   4: <T as wgpu::context::DynContext>::surface_configure
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-0.15.1/src/context.rs:2053:9
   5: wgpu::Surface::configure
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/wgpu-0.15.1/src/lib.rs:4113:9
   6: egui_wgpu::winit::Painter::configure_surface
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/egui-wgpu-0.21.0/src/winit.rs:136:9
   7: egui_wgpu::winit::Painter::resize_and_generate_depth_texture_view
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/egui-wgpu-0.21.0/src/winit.rs:245:9
   8: egui_wgpu::winit::Painter::set_window::{{closure}}
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/egui-wgpu-0.21.0/src/winit.rs:214:17
   9: pollster::block_on
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/pollster-0.3.0/src/lib.rs:128:15
  10: eframe::native::run::wgpu_integration::WgpuWinitApp::init_run_state
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/eframe-0.21.3/src/native/run.rs:1114:17
  11: <eframe::native::run::wgpu_integration::WgpuWinitApp as eframe::native::run::WinitApp>::on_event
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/eframe-0.21.3/src/native/run.rs:1302:25
  12: eframe::native::run::run_and_return::{{closure}}
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/eframe-0.21.3/src/native/run.rs:161:28
  13: winit::platform_impl::platform::wayland::event_loop::EventLoop<T>::run_return
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.28.1/src/platform_impl/linux/wayland/event_loop/mod.rs:241:9
  14: winit::platform_impl::platform::EventLoop<T>::run_return
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.28.1/src/platform_impl/linux/mod.rs:785:56
  15: <winit::event_loop::EventLoop<T> as winit::platform::run_return::EventLoopExtRunReturn>::run_return
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/winit-0.28.1/src/platform/run_return.rs:51:9
  16: eframe::native::run::run_and_return
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/eframe-0.21.3/src/native/run.rs:124:5
  17: eframe::native::run::wgpu_integration::run_wgpu::{{closure}}
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/eframe-0.21.3/src/native/run.rs:1410:17
  18: eframe::native::run::with_event_loop::{{closure}}
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/eframe-0.21.3/src/native/run.rs:108:9
  19: std::thread::local::LocalKey<T>::try_with
             at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/std/src/thread/local.rs:446:16
  20: std::thread::local::LocalKey<T>::with
             at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/std/src/thread/local.rs:422:9
  21: eframe::native::run::with_event_loop
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/eframe-0.21.3/src/native/run.rs:101:5
  22: eframe::native::run::wgpu_integration::run_wgpu
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/eframe-0.21.3/src/native/run.rs:1407:13
  23: eframe::run_native
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/eframe-0.21.3/src/lib.rs:207:13
  24: re_viewer::native::run_native_app
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/re_viewer-0.2.0/src/native.rs:32:5
  25: re_viewer::native::run_native_viewer_with_messages
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/re_viewer-0.2.0/src/native.rs:50:5
  26: re_sdk::session::Session::show
             at /home/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/re_sdk-0.2.0/src/session.rs:397:9
  27: test_rerun::main
             at ./src/main.rs:26:5
  28: core::ops::function::FnOnce::call_once
             at /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/ops/function.rs:507:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

jleibs, 2023-02-16:
This looks very similar to some other issues with wayland and wgpu, I've seen. Could you maybe try running through X11 / xwayland instead?
```
export WINIT_UNIX_BACKEND=x11
```

andrsbtrg, 2023-02-17:
that was it. Now I was able to (re)run . Going now through the docs. Thanks a lot!

closed 2023-02-17 by andrsbtrg