rerun-io/rerun #1343: Publish crates on CI

emilk, 2023-02-17:
We still want to be able to publish locally (in case of emergency), but the default should be that the crates are published when a new release-tag is pushed.

emilk, 2023-05-02:
We have a `scripts/publish_crates.sh` script which could perhaps be replaced with a 3rd party helper that finds all the crates that should be published, and publish them in a topological order.

There are things like `cargo-release` that does this and more, and we could perhaps use that if we could get rid of the "and more" part :)

emilk, 2023-05-03:
We could for-loop-publish all our crates until they all succeed. No need for topological sort. Could be a simple python script.

jleibs, 2023-05-25:
Another idea, similar to how we handle the wheel files, would be to publish these crates initially to gcloud. That would allow us to use the final version numbers during the testing phase and then only re-publish them to crates.io when we're happy with them.

jprochazk, 2023-06-29:
- [ ] add as many checks from `cargo publish` to our checks workflows as possible (probably can't match it exactly- that's fine)
- [ ] use `cargo package` on the crates and publish them to `build.rerun.io` for every commit (`build.rerun.io` becomes our own miny-registry)
- [ ] publish crates in the `manual_release` workflow

jprochazk, 2023-07-18:
Finished in https://github.com/rerun-io/rerun/pull/2721

closed 2023-07-18 by jprochazk