rerun-io/rerun #1345: Keeping our docs up-to-date

emilk, 2023-02-17:
A few things we should consider adding:

* link checker to `rerun` and `rerun-docs` repos
* add a CI action that reminds us to update our docs whenever we add the API label
* Use [`cargo-semver-checks`](https://crates.io/crates/cargo-semver-checks) to check API changes
* Use some tool to look for Python API changes
* Use an IPython notebook for our getting-started guides, so we can make sure it compiles

jleibs, 2023-03-08:
One thing we should make sure this covers is running link-check on the latest rerun-docs when we change the rerun repo. We have a few places where we cross-link from rerun-docs to github and so it's possible that if we move things around in the repo we are breaking existing links somewhere else.

obi1kenobi, 2024-02-13:
👋 Hi! I'm the author of `cargo-semver-checks` and I just stumbled across this issue. I noticed the `RELEASES.md` file seems to indicate you're already using `cargo-semver-checks` — that's awesome. Has it been working well so far?

Are you still looking for a Python API semver linter? I'm about to release one based on the same underlying technology as cargo-semver-checks, and I'm looking for a few early-adopters to test it out in a small private beta. Let me know if you might be interested!