rerun-io/rerun #1361: Support for more 3D primitives

jleibs, 2023-02-20:
It would be nice to be able to log the common 2d/3d geometry building blocks in order to enable building up simple 3d models without going all the way to fully specified meshes.
 - Rect / Box
 - Circle / Sphere
 - Triangle / Cone
 - Cylinder

Maybe more, but [good enough for VRML, good enough for me?](https://web.cs.wpi.edu/~matt/courses/cs563/talks/vrml/primitive.html)

These should all consistently support components for Orientation, Color, and Fill.

emilk, 2023-04-27:
I think we should add a minimal version of this ASAP - it can be something that just generates meshes during scene construction for now. This is very limiting at the moment, and we need this to test other things like transparency, 2D layering, and styling.

emilk, 2023-05-01:
An interesting note: a `Rectangle` component is useful both in 2D and 3D.

For 2D we support it via specifying its min/max corners, or center and size, and some other options. We will also add support for 2D transforms (https://github.com/rerun-io/rerun/issues/349).

For 3D we may want the same 2D interface plus a 3D transform, treating the 2D rect as a 3D shape with an implicit Z=0.

emilk, 2023-05-03:
Rect/Box we have, but needs styling (Separate issue).

Cylinders and capsules I think are very useful (e.g. limbs)

2D circles/discs are very useful. So are spheres. What is the difference with point?

Point radii can be auto and ui-based, but spheres/circles _need_ a real radius.
A sphere/circle would squash with non-uniform transforms.
Points are using a faster, less accurate, renderer.

jeongseok-meta, 2024-01-18:
It would be great to have a way to visualize the grid or ground. This could be achieved through a loggable object or simply as a visual artifact.

mont1004, 2024-04-08:
Hi there. 

Just curious, has there been any progress or updates on this topic? Would love to know what's going on.
I personally hope for support of an ellipsoid representation in physical units, which can be beneficial for expressing the ambiguity in, for example, camera pose estimation.

abey79, 2024-04-08:
@mont1004 Unfortunately the next really will have little new on this front, but this is still something that's high on our priority list. Do you have any particular primitive in mind that you would need most?

mont1004, 2024-04-08:
@abey79 Thank you for the quick reply!

My team has been developing Visual SLAM, and we've been eyeing the rerun as a top candidate for our new visualizer. As I mentioned earlier, we currently represent the covariance of the estimated camera poses with ellipsoids. Having ellipsoids that can be scaled and compared in size with one another would be very helpful, as this capability is crucial for us.

pablovela5620, 2024-04-26:
Want to add like #6012 that adding text to an image, similar to OpenCV would be appreciated, oftentimes I don't want to log something in textlog/textdocument as I would like it to show in the image itself. As a work around I log a point2d or linestrip with 0 radius and add a label. This works, but I can't easily change the size of the font like cv2.drawtext would allow

nikolausWest, 2024-04-26:
@pablovela5620: Are you adding the text to a particular place in the image (i.e. tying it to a geometric location), or is the text more of a description of the image (i.e. it could alternatively be below the image for instance)?

pablovela5620, 2024-04-26:
Generally to a geometric location, for example detecting someone's face and placing the text next to or above their detected face. Then telling them to either get closer or farther from the screen or to move their face to a specific location. Below an image could also be helpful (though I think this would already be possible with a textdoc + blueprint?)

nikolausWest, 2024-06-10:
Rays and panes as per #2179 would be great

emilk, 2024-06-10:
For the 3D primitives, we should have a style option to either render them as solid, as wireframe, or both. The spatial space-view can convert them to a re_renderer mesh and/or lines as needed.

To keep things simple, I suggest we use the `Transform3D ` archetype to orient and position the shapes. This means that e.g. a cylinder would always have its base at the origin, and be defined by `Cylinder { radius, height }`.

Which axis should "height" be along? I doesn't really matter, but once we picked something we should stick with it forever, and also use it for all other shapes (i.e. the axis along which a capsule and cylinder stretches). It's worth spending some time investigating what is the most common choice in other visualization tools.

We should still have nice helpers to construct these primitives. For instance, defining a capsule by two points and a radius (a fat line segment) but then logging that as a single `component Capsule { radius, length }` plus an `Transform3D`. This would mean an `archetype Capsule` would also _contain_ `archetype.Transform3D` in a sense, which may require some more codegen design/thinking.

For the 3D ellipsoids: I did some preliminary experimentation with these in https://github.com/rerun-io/rerun/pull/4991

teh-cmc, 2024-06-10:
> We should still have nice helpers to construct these primitives. For instance, defining a capsule by two points and a radius (a fat line segment) but then logging that as a single `component Capsule [...]

We use the `OutOfTreeTransform3D` component for this today; it has worked well for `Asset3D`.

emilk, 2024-07-01:
A less important, but very nice, primitive, is the tapered capsule:

[image]

kpreid, 2024-07-24:
Progress update:

* #6853 has added the `Ellipsoids` archetype.
* #6953 has added general mechanisms for solid/filled surface rendering (so our 3D geometric shapes can be solid) and applied that to `Boxes3D` and `Ellipsoids`.

More 3D shapes to come!

emilk, 2024-10-28:
# What is the main axis of a cylinder/capule?
URDF: ???
## X
* PhysX: https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/Geometry.html#capsules

## Y Main axis
* VRML: https://www.wiley.com/legacy/compbooks/vrml2sbk/ch03/03fig05.htm
* three.js: https://github.com/mrdoob/three.js/blob/master/src/geometries/CylinderGeometry.js

emilk, 2024-10-28:
A lot of 3D libraries have a configurable main-axis, that the user can set to any Vec3. I think we should provide the same. That way a user can easily configure where a cylinder points, without having to calculate a quaternion and read our docs about what our default is.

So I suggest we introduce a `components.MainAxis3: datatypes.Vec3` that defaults to `+Z`.
A user can still add any transform on top of it (including rotation).

0x7CFE, 2025-10-03:
Hello,

I'd like to contribute by implementing `Circles2D` primitive since I need it to visualize my ML model.

Currently I'm using an app I wrote in [iced](https://github.com/iced-rs/iced) where it all looks something like this (the app in action can be seen on [this Youtube video](https://youtu.be/n27tv8KyUDY)):
[image]

I was able to use Points2D for the top-left view, but the remaining ones are tricky. Since `Circles2D` is probably not a top priority, I am wondering, how hard it is to implement? If it's not that hard maybe I would help this awesome project by implementing the primitive myself.

abey79, 2025-10-03:
@0x7CFE thanks for asking.

To be honest, it is a surprisingly involved task, no so much because of the actual rendering (this could be done e.g. by piggy-backing on the existing `LineStrip2D` rendering, though there might be better ways), but because the features is about introducing a new piece in Rerun's data model. That includes discussing/designing the archetype, setting up codegen for the various SDKs, producing example snippets, etc. It also involves ensuring that all the relevant views (in this case, the spatial 2d and map view) add support for the new archetype. You can have a taste of that by looking at some of the PRs referenced above.

That being said, it is certainly not an impossible task, and is mainly about retracing existing pathways in the code base. I don't mean to discourage a contribution here, but just set the expectations straight :)

abey79, 2025-10-15:
Very cool visual! 🤩 

Yeah alpha is currently not supported for points.

Reusing `FillMode` for 2d shapes seems slightly weird to me. However, we could add a `background_colors` optional component. (Having two components of the same type was impossible until recently, but this restriction has now been lifted).

I'm not sure the rendering side is ready for 2d shape transparency support, but an initial version could already support either 255 or 0 alpha to control the presence/absence of the background. It's also possible that some works need to be done on the rendering side for tessellation.

But @Wumpf should fact check all of the above, as I've been slightly out of touch with these parts recently.

Wumpf, 2025-10-15:
Yeah, we haven't figure out our 2d draw order story yet which is why we don't have transparency on 2D things yet. It got recently unblocked since the renderer finally performs distance based sorting, but 2D draw order is still controlled via a fairly flaky z-buffer hack.
Doing a 255/0 hack isn't great imho since "not drawing" has very different semantics also for selection. I'd rather "just fix it finally" ;-)

And yes, fact checking that this screenshot looks pretty cool ✔️ 😁

0x7CFE, 2025-10-15:
Oh, I see. Honestly, `background_colors` would work perfectly fine for me. All things considered, what would be the proper way to move forward?

P.S.: It looks even better after the clustering is solved 😁
[image]