rerun-io/rerun #1373: Memory use: Use ref-counted buffers for images and meshes

emilk, 2023-02-21:
When we load a log message in the store we clone their contents.

Currently we use `Vec` when storing the contents of an image (`TensorData::U8`) and a mesh (`RawMesh3D`).

We want to use ref-counted types like `Arc` or `arrow2::Buffer`, otherwise we duplicate our memory use.

The natural thing to use is `arrow2::Buffer<u8>` or `Arc<[u8]`, but neither of those implement `ArrowSerialize/ArrowDeserialize` yet (see below).

See:
* https://github.com/rerun-io/rerun/pull/1372
* https://github.com/DataEngineeringLabs/arrow2-convert/issues/101
* https://github.com/DataEngineeringLabs/arrow2-convert/issues/102

nikolausWest, 2023-02-22:
While I definitely agree with these improvements I think this issue also points to a higher level question: should we really be keeping all the incoming log messages in memory at all? 

First of all we should probably make it easier to set up ways to stream messages directly to a file, both from the SDK and as it hits the viewer / server. 

Still, it's very useful to be able to decide to save a recording (`.rrd` file) after the fact. Shouldn't we be able to to quite simply serialize and deserialize the data store to achieve the same ends?

teh-cmc, 2023-04-18:
Fixed:
- We don't store data `LogMsg`s anymore.
- `TensorData`, `EndcodedMesh3D` and `RawMesh3D` all use `Buffer`s under the hood.

closed 2023-04-18 by teh-cmc