rerun-io/rerun #1387: view_coordinates for 2D views

jleibs, 2023-02-23:
View_coordinates are currently ignored for 2d views. Should be possible to use these to change the way x and y are mapped to up / right on the screen.

Related to: https://github.com/rerun-io/rerun/issues/349

emilk, 2023-03-20:
Blocked on replacing our `egui::Image` with re_renderer

Wumpf, 2023-04-18:
I took a quite successful stab at #1581 which just like this ticket requires touching the camera we use for 2D views, making this related

emilk, 2023-05-03:
This is pretty important when plotting something arbitrary and you want Y=up

patrickelectric, 2024-02-16:
Any plans to have a open street map visualizer or something like it ?

Wumpf, 2024-02-18:
@patrickelectric we're hoping it's soon possible to log all the necessary 2d primitives to rerun to make it possible to write an application that puts openstreet map data into Rerun, but we haven't planned to do something like that third party. That's offtopic though - this issue is about a very particular aspect of the 2d view

emilk, 2024-04-12:
We should do this first:
* https://github.com/rerun-io/rerun/issues/2663