rerun-io/rerun #1397: Turn off store archiving to maybe solve memory duplication issue

emilk, 2023-02-24:


emilk, 2023-02-27:
The way to accomplish this is to set

```
DataStoreConfig {
    component_bucket_size_bytes: u64::MAX,
    component_bucket_nb_rows: u64::MAX,
    …
}
```

but according to the docs, that stops the GC from doing anything.

However, considering we don't really have a working GC right now, maybe it doesn't matter:
* https://github.com/rerun-io/rerun/issues/1414
* https://github.com/rerun-io/rerun/issues/1412

emilk, 2023-03-03:
No - we want memory pruning

teh-cmc, 2023-03-09:
This turned out to be the right strategy in the end.

It need not break garbage collection: we can disable archiving without necessarily breaking the bucketing system.

PR #1535 (2023-03-09): datastore: disable compaction (fixes 2x memory issue)
Disables bucket-level compaction aka. archiving until we support batching, thereby fixing the 2x memory issue. See below for rationale.

This doesn't impact garbage collection nor performance in any noticeable way.

Before:
```
$ cargo r -p test_image_memory
Logged 100 2048x1024 RGBA images = 800 MiB
1.6 GiB RAM used
```

After:
```
$ cargo r -p test_image_memory
Logged 100 2048x1024 RGBA images = 800 MiB
831 MiB RAM used

```

Closes #1397 

---

The original long form rationale:
> Alright so I've continued digging down the LogMsg-less rab [... cut]
closed 2023-03-09 by teh-cmc