rerun-io/rerun #1399: Save-file dialog hangs rerun on Ubuntu 22.04

jleibs, 2023-02-24:
Looks like maybe a missing dependency or a version mismatch issue:

Rerun works fine until trying to save an RRD file at which point it hangs and we get glib errors in the console.

```
(python:74188): GLib-GObject-WARNING **: 09:55:43.734: cannot register existing type 'GSettingsBackend'

(python:74188): GLib-GObject-WARNING **: 09:55:43.734: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(python:74188): GLib-CRITICAL **: 09:55:43.734: g_once_init_leave: assertion 'result != 0' failed

(python:74188): GLib-GObject-CRITICAL **: 09:55:43.734: g_type_register_static: assertion 'parent_type > 0' failed

(python:74188): GLib-CRITICAL **: 09:55:43.734: g_once_init_leave: assertion 'result != 0' failed

(python:74188): GLib-GIO-WARNING **: 09:55:43.734: Tried to implement non-registered extension point gsettings-backend

(python:74188): GLib-GObject-WARNING **: 09:55:43.734: cannot register existing type 'GProxyResolver'

(python:74188): GLib-GObject-CRITICAL **: 09:55:43.734: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(python:74188): GLib-CRITICAL **: 09:55:43.734: g_once_init_leave: assertion 'result != 0' failed

(python:74188): GLib-GObject-CRITICAL **: 09:55:43.734: g_type_add_interface_dynamic: assertion 'g_type_parent (interface_type) == G_TYPE_INTERFACE' failed

(python:74188): GLib-GIO-WARNING **: 09:55:43.734: Tried to implement non-registered extension point gio-proxy-resolver
```

Wumpf, 2023-02-24:
might be related to https://github.com/rerun-io/rerun/issues/1315

jleibs, 2023-02-24:
Seems like a dupe of https://github.com/rerun-io/rerun/issues/1315. Will close and put further investigation details there.