rerun-io/rerun #1404: Send analytics on crash

emilk, 2023-02-25:
We should send an event on `panic` and `SIGSEGV`.

We should NOT include the panic message, as that could include file paths and other personal information.

We should however include the backtrace, as will be anonymous and useful.

Wumpf, 2023-02-26:
> We should however include the backtrace, as will be anonymous and useful.

I'm a little bit nervous about this assessment. That's true for the regular usecase of connecting to the viewer from another process, but if the viewer itself is used as a library and spawned in the same process as other user-code, the callstack can be very revealing. This probably isn't much of an issue right now, but once we think about extensibility it will be.

emilk, 2023-02-26:
We’ll only install the panic/signal handler in our binary, not in any library

closed 2023-02-28 by emilk