rerun-io/rerun #1427: Create wgpu::RenderPipelines just-in-time to avoid compiling shaders for unused pipelines

Wumpf, 2023-02-28:
As re_renderer gets more complex we get more and more variations of render pipelines. Since currently we need a mutable context for creation pipelines and have only an immutable one for rendering, a lot of the creation happens "in case" it's needed. Solving this is desirable, but arguably scratches only the surface and might lead to the same actual issue - creating of unneeded renderpipelines - down the line.

So what we really want is to create the actual resource whenever re_renderer code is about to _bind_ a renderpipeline, i.e. resolves the handle to the underlying resource!
I believe our architecture is in a good state for this, shouldn't be too hard to defer creation for *all* static resource pools!

This is not much of a concern right now, but going forward we don't want to pay the cost for e.g. volumetric rendering pipelines when all the user does is look at a picture - startup times should not suffer from potential features!

teh-cmc, 2023-02-28:
That sounds like a dupe of #701?

Wumpf, 2023-03-01:
It is! Forgot that I already did a writeup on this

closed 2023-03-01 by Wumpf