rerun-io/rerun #1460: Metric units & mapping metric scales to world-space units

teh-cmc, 2023-03-01:
Had a long discussion with @Wumpf about all things size & scale and one of the biggest issue right now is that for most things we render, we don't have any notion of what their actual size is in terms of the metric system, and even then we have no way of specifying how a metric unit actually maps to a world-space unit in the scene (e.g. 1 world-space unit = 1mm).

Obviously this becomes a real issue as soon as there's more than one thing to render in a scene :shrug:.

Some (few) of our primitives actually already offer a way to specify their metric scale: e.g. users can specify a metric scale when logging depth maps.
Unfortunately, when projecting said depth maps into point clouds in a scene, there's no way of telling how that size relates to the rest of the scene.

We should have a way of specifying a metric scale for most things, and then have a way of specifying the correspondence from metric units to world-space units when setting up a view.

emilk, 2023-03-01:
The issue as written seems to be vaguely touch three different things:

A) There is sometimes a desire to somehow specify what the unit of measurement is per “space”. Is `1` a meter, a lightyear, or a pixel? This doesn’t affect anything except, perhaps, how we present distances in the GUI.

B) There is the problem of specifying scale in transforms. For instance: transforming a value in a depth-map tensor to a distance in the parent space.

C) It is sometimes useful to be able to say “this is the rough scale of things in this space”. Two users can both be using `mm` as their base unit, but one is 3D scanning whole cities, and the other is 3D scanning jewelry. This “general scale of things” can often be estimated from the actual logged data.

teh-cmc, 2023-03-01:
Another thing: I want to be able to configure my camera speed as e.g. `10 m/s`, which would just do the right thing based on how big I've defined a world unit to be in my current view.