rerun-io/rerun #1470: Github latest release page connected to wrong commit-hash

jleibs, 2023-03-02:
https://github.com/rerun-io/rerun/releases/tag/latest now has just the artifacts from the most recent build, but it's still connected to a stale hash.

This means it incorrectly shows "84 commits since this release"
[image]

And also the attached Source archive is incorrect.

jleibs, 2023-03-02:
I'm not seeing an option to support this in: https://github.com/ncipollo/release-action

We may need to just use a separate action to manually update the git tag itself.

emilk, 2023-03-02:
It's weird that `latest` has the artifacts from the most recent build - `latest` is the tag we use for proper releases

closed 2023-03-02 by jleibs