rerun-io/rerun #1480: UI freezes when trying to save

tlpss, 2023-03-02:
**Describe the bug**
Rerun UI freezes without succesfully saving when I try to save to a file.

**To Reproduce**
Steps to reproduce the behavior:
1. run `python -m rerun`
2. log some data to rerun 
3. try to save using the UI

**Expected behavior**
Expect Rerun UI to save a datafile and then continue working.

**Screenshots**
[image]

**Backtrace**
```
$ python -m rerun
2023-03-02T14:22:25.684096Z  INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 1    

(python:940210): GLib-GObject-WARNING **: 15:26:17.681: cannot register existing type 'GVfs'

(python:940210): GLib-GObject-WARNING **: 15:26:17.682: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(python:940210): GLib-CRITICAL **: 15:26:17.682: g_once_init_leave: assertion 'result != 0' failed

(python:940210): GLib-GObject-CRITICAL **: 15:26:17.682: g_type_register_dynamic: assertion 'parent_type > 0' failed

(python:940210): GLib-GObject-WARNING **: 15:26:17.682: cannot register existing type 'GVolumeMonitor'

(python:940210): GLib-CRITICAL **: 15:26:17.682: g_once_init_leave: assertion 'result != 0' failed

(python:940210): GLib-GObject-CRITICAL **: 15:26:17.682: g_type_register_dynamic: assertion 'parent_type > 0' failed

(python:940210): GLib-GIO-WARNING **: 15:26:17.682: Tried to implement non-registered extension point gio-volume-monitor

(python:940210): GLib-GObject-WARNING **: 15:26:17.682: cannot register existing type 'GLoadableIcon'

(python:940210): GLib-GObject-WARNING **: 15:26:17.682: cannot register existing type 'GIcon'

(python:940210): GLib-GObject-CRITICAL **: 15:26:17.682: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(python:940210): GLib-CRITICAL **: 15:26:17.682: g_once_init_leave: assertion 'result != 0' failed

(python:940210): GLib-CRITICAL **: 15:26:17.682: g_once_init_leave: assertion 'result != 0' failed

```
**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04 Desktop

**Additional information**
rerun version: `0.2.0+df920dc.1`. Installed precompiled wheel from release pages since a new wheel has not been uploaded to PyPI yet afaik.

teh-cmc, 2023-03-02:
(Dupe of #1315 & #1399, will very likely be fixed by #1402)

emilk, 2023-03-02:
Closed as duplicate of https://github.com/rerun-io/rerun/issues/1315