rerun-io/rerun #1496: Cannot add Pinhole component, rerun::Component not implemented for Pinhole

EriKWDev, 2023-03-03:
**Describe the bug**
I'm trying to add a `rerun::components::Pinhole` to a message, but when I add the Pinhole component to a message, I get a compilation error
```console
    |
    |       .with_component(&[pinhole_projection])
    |        ^^^^^^^^^^^^^^ the trait `rerun::Component` is not implemented for `rerun::components::Pinhole`
    |
```
```console
    |
183 |     pub fn with_component<'a, C: SerializableComponent>(
    |                                  ^^^^^^^^^^^^^^^^^^^^^ required by this bound in `MsgSender::with_component`

```

**To Reproduce**
Steps to reproduce the behavior:
Create a MsgSender with a Tensor from_image as a component. When you add a rerun::components::Pinhole to it it says that Pinhole is not a component

```rust

let pinhole_projection = rerun::components::Pinhole {
    image_from_cam: [
        [1.0, 0.0, 0.0],
        [0.0, 1.0, 0.0],
        [0.0, 0.0, 1.0],
    ]
    .into(),
    resolution: Some([1920.0, 1080.0].into()),
};

rerun::MsgSender::new("Image")
    .with_component(&[rerun::components::Tensor::from_image(image).unwrap()])
    .unwrap()
    .with_component(&[pinhole_projection])
    .unwrap()
    .send(&mut rr_session)
    .unwrap();

```

**Expected behavior**
Ability to add Pinhole component like any other components

**Desktop:**
 - OS: Linux, Debian bullseye, Gnome version 3.38.5

**Additional context**
```console
> cargo --version
cargo 1.67.0 (8ecd4f20a 2023-01-10)
> rustc --version
rustc 1.67.0 (fc594f156 2023-01-24)
```
Running on rerun commit `27ccf7aa9d20d2216f222465cae57f48cf304057` (HEAD as of writing)

EriKWDev, 2023-03-03:
Apparently this is perhaps not as much a bug as an ergonomics/docs thing. I found now that it should be added using `rerun::components::Transform::Pinhole(pinhole_projection)`

```rust
rerun::MsgSender::new("Image")
    .with_component(&[rerun::components::Tensor::from_image(image).unwrap()])
    .unwrap()
    .with_component(&[rerun::components::Transform::Pinhole(
        pinhole_projection,
    )])
    .unwrap()
    .send(&mut rr_session)
    .unwrap();
```

Maybe Pinhole shouldn't be in the `rerun::components` module?

nikolausWest, 2023-09-11:
This is fixed now with updated api landing in 0.9

closed 2023-09-11 by nikolausWest