rerun-io/rerun #1510: Configurable ranges / extent in image and tensor view

tfachmann, 2023-03-05:
I was fiddling around with rerun and see large potential to use it for data analysis in our field where I would like to replace matplotlib by rerun. Selecting dimensions to slice through is pretty awesome!

[image]

To actually use it (i.e. extracting coordinates intuitively, slicing through the actual dimension) I would like to have an option to add an *extent* (or axis limits) similar to matplotlib: https://matplotlib.org/stable/tutorials/intermediate/imshow_extent.html

The option to add axes and a grid would help as well.

Note: the data I'm visualizing here is 3D (X, Y, Z in meter) with some quality.

It is related to #683

nikolausWest, 2023-03-05:
Hey @tfachmann! Thanks for opening this issue and really fantastic to hear you see a large potential in Rerun!

It's almost funny that this issue came in during this weekend because on Friday, we aligned internally around that during the next 6 weeks, Rerun should become a completely valid and significantly better alternative to `imshow`. The first week or two will be spent mostly on laying the ground work so I can't promise when this particular issue will be solved, but at least know it's a top priority for us right now.

tfachmann, 2023-03-06:
Cool, nice to hear it aligns with what you're trying tackle anyway! Looking forward for upcoming releases and can't wait to show it in our institute 🚀

emilk, 2023-03-11:
So what you are asking for is basically a cropping of the tensor, correct?

Do you want/need this to be precise, or do you really just want to zoom in to the tensor?

tfachmann, 2023-03-11:
In my case, not really (it may be that it is not 100% `extent` what I'm looking for). I basically want to give real world coordinates to my data (per axis). E.g. I know that my X axis starts at -1.5 meter and ends at 1.2 meter. For hovering /  slicing and other operations I would expect rerun to show the respective values of the given Dimension instead of simple ranges e.g. 0-250 that I must calculate back manually

emilk, 2023-03-11:
Ah I see - you want to map the tensor indices to some other coordinate system. This could perhaps be handled by the `transform` system then.

Thanks for the clarification!