rerun-io/rerun #1511: Rerun as a better alternative to imshow

nikolausWest, 2023-03-05:
Rerun should become a valid replacement and obvious improvement to using `cv2.imshow` or `plt.imshow`.

This requires (at least):

- [ ] #1510
- [ ] Optionally show grid on image and tensor views
- [ ] Optionally show axis on image and tensor views
- [ ] Rerun renders inline in Jupyter Notebooks
- [ ] Zoomed in value inspection on hover for tensor views
- [ ] Images in 2D views are special cases of tensor views
- [ ] All image and tensor related blueprint options are settable from the Python and Rust SDKs

Erol444, 2023-03-06:
To add to this the list, **disabling logging** would likely be required as well for runtime applications as well, as otherwise, users would quickly run out of RAM:)

nikolausWest, 2023-03-06:
@Erol444: Do you mean just setting the logging SDK to be no-op? That can be done already. See the parameter `default_enabled` in the init method: https://ref.rerun.io/docs/python/v0.2.0/common/initialization/#rerun.init

Erol444, 2023-03-06:
@nikolausWest Is this working as expected, for Python as well? Similar to [this issue](https://github.com/rerun-io/rerun/issues/1412) (& my comments there), it looks like it isn't working, or perhaps there's a memory leak somewhere else. I tried it with `default_enabled=False`, but it seems like it's still logging:
[image]

teh-cmc, 2023-03-06:
Can confirm `default_enabled` is broken in the python bindings at the moment; will push a fix in a moment so that it ships as part of the imminent `0.3.0` release.

In the meantime you can use `rr.set_enabled` to achieve the same effect, that one still works.

nikolausWest, 2023-03-08:
@Erol444: could you give a little more details on what you are trying to achieve? I think I may have originally misinterpreted  what you meant by disabling logging.

Erol444, 2023-03-08:
Hi @nikolausWest , I just want the live-preview of the image/pointcloud/... without actually saving the history (logging?) by writing these objects into RAM, as otherwise it fills up the whole RAM and slows down. I hope that makes sense!

nikolausWest, 2023-03-09:
> Hi @nikolausWest , I just want the live-preview of the image/pointcloud/... without actually saving the history (logging?) by writing these objects into RAM, as otherwise it fills up the whole RAM and slows down. I hope that makes sense!

Ah, alright, I understand. We use the name logging for the operation that takes data out of your program and into Rerun (whether that is a .rrd file or the Viewer). That it fills up RAM is based on two separate problems:

- #1242 
-  #1539

We're working on fixing both of these right now.

nikolausWest, 2023-03-09:
Not quite a requirement for replacing `imshow` but this still feels related:
- #1544

nikolausWest, 2023-03-20:
Superseded by #1630

closed 2023-03-20 by nikolausWest