rerun-io/rerun #1520: Rerun catches SIGINT and treats it as a crash when serving the web viewer

teh-cmc, 2023-03-06:
Which of course implies a crash report sent to posthog and everything.

```
$ rerun --web-viewer examples/out/api_demo_a1.rrd

2023-03-06T22:38:42.722126Z  INFO rerun::run: Loading "examples/out/api_demo_a1.rrd"…
2023-03-06T22:38:42.722206Z  INFO re_ws_comms::server: Listening for websocket traffic on 0.0.0.0:9877. Connect with a web Rerun Viewer.
^C

Rerun caught a signal: SIGINT

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting

   2: <unknown>
   3: syscall
   4: std::sys::unix::futex::futex_wait
             at std/src/sys/unix/futex.rs:62:21
      std::sys::unix::locks::futex_condvar::Condvar::wait_optional_timeout
             at std/src/sys/unix/locks/futex_condvar.rs:49:17
      std::sys::unix::locks::futex_condvar::Condvar::wait
             at std/src/sys/unix/locks/futex_condvar.rs:33:9
   5: tokio::runtime::park::Inner::park
   6: tokio::runtime::park::CachedParkThread::block_on
   7: tokio::runtime::scheduler::multi_thread::MultiThread::block_on
   8: rerun::main
   9:
Rerun caught a signal: SIGINT
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
```

```
$ python -m rerun --web-viewer examples/out/api_demo_a1.rrd

2023-03-06T22:41:16.014647Z  INFO rerun::run: Loading "examples/out/api_demo_a1.rrd"…
2023-03-06T22:41:16.014733Z  INFO re_ws_comms::server: Listening for websocket traffic on 0.0.0.0:9877. Connect with a web Rerun Viewer.
^C

Rerun caught a signal: SIGINT

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting

   2: <unknown>
   3: syscall
   4: std::sys::unix::futex::futex_wait
             at std/src/sys/unix/futex.rs:62:21
      std::sys::unix::locks::futex_condvar::Condvar::wait_optional_timeout
             at std/src/sys/unix/locks/futex_condvar.rs:49:17
      std::sys::unix::locks::futex_condvar::Condvar::wait
             at std/src/sys/unix/locks/futex_condvar.rs:33:9
   5: tokio::runtime::park::Inner::park
[...]
  25: Py_RunMain
  26: Py_BytesMain
  27: <unknown>
  28: __libc_start_main
  29: _start

Rerun caught a signal: SIGINT
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
```

Wumpf, 2023-03-06:
I think this is the same as  #1367 (and #1312? I'm confused about these)? If yes, close this old one, this here is more expressive

teh-cmc, 2023-03-07:
I don't think these are the same thing? These previous tickets were about the fact that it's impossible to kill the web-server... but clearly now we can (so this got fixed at some point..?!), but now this gets caught as a crash :|

I'll go ahead and add this ticket to the current cycle, this seems like one of these uber annoying subject matter that could benefit from all of us being in the same room.

closed 2023-03-21 by Wumpf