rerun-io/rerun #1531: Log an image explicitly as a 3D surface/billboard

jleibs, 2023-03-07:
Motivating use-case: I have an image that represents a map. I want to place that map on the "floor" of my 3D space. This is different from a camera projection because I want the image plane to be located at specific geometric location with a specified mapping from pixels to 3d distance.

Example of equivalent functionality in rviz:
[image]

This seems like it could mostly be handled with a Rigid3 transform + scale to map from pixel -> spatial coordinates.

Wumpf, 2023-03-07:
On the re_renderer side we support arbitrary spanning vectors for textured rectangles. This means that this is mostly a matter of re_viewer and SDK :)

tim-fan, 2025-01-09:
Adding some keywords in case others search for this issue - I almost logged a duplicate

ROS, OccupancyGrid, costmap, map

nickcharron, 2025-09-28:
Do we have a solution for this yet @jleibs @tim-fan ?

I tried to do this by just logging an image and a Transform3D. The image appears, but not in the 3D viewer. Here's a minimal example:

```
checkerboard = np.array([[(i+j)%2 for i in range(10)] for j in range(10)], dtype=np.uint8) * 255
checkerboard = np.kron(checkerboard, np.ones((20,20))).reshape(200, 200, 1)
debug_image = np.repeat(checkerboard, 3, axis=2)

# Log it to the world frame
rr.log("world", rr.ViewCoordinates.RIGHT_HAND_Z_UP, static=True)
rr.log(
    "world/checkerboard",
    rr.Transform3D(
        translation=[0, 0, 0.5],
        scale=[10, 10, 1.0],
        rotation=rr.Quaternion(xyzw=[0, 0, 0, 1])
    )
)
rr.log("world/checkerboard", rr.Image(debug_image))
```

tim-fan, 2025-09-28:
I haven't tried yet, but I was thinking to log some pinhole extrinsics to get the image showing in 3d

https://rerun.io/docs/reference/types/archetypes/pinhole

If I'm correct, the example above projects an image to a  3x3m grid? So should be able to modify for any arbitrary cell size and resolution? That was my line of thinking, haven't tried it yet.

iwanders, 2025-10-25:
Hi all, the mesh solution proposed by @nickcharron looks good, unfortunately, it wasn't performant enough for the grid I needed to visualise (~20e6 pixels), a mesh that large resulted in 100% gpu usage and an unusable framerate.

Instead, what I ended up doing is calculating a pinhole transform and using the image_plane_distance property to position the image at the correct location and scale. This has excellent performance. By placing the pinhole origin far away the cursor rays are pretty much perpendicular to the ground, which is also helpful if you combine the 2d view with the 3d view. In the viewer, it is then possible to make pinhole's `Camera->line_width` 0 pixels; this hides the pinhole frustrum. Unchecking `Interactive` on the camera/grid entity ensures it behaves nicely as an unobtrusive map display.

Python function:
```py
def log_grid_xy(
    grid,
    prefix="prefix",
    resolution=0.05,
    bottom_left_origin=(0.0, 0.0),
    image_plane_distance=100.0,
    grid_z=0.0,
):
    """
    Log an image as a grid in the xy plane with pixels of resolution size.

    Workaround for https://github.com/rerun-io/rerun/issues/1531

    Parameters
    ----------
    grid : numpy.ndarray
        The numpy array to display as grid, passed to rr.Image. May be 3 channels for rgb.
    prefix : str
        Prefix to use for the entity.
    resolution: float
        Size of a grid pixel in world coordinates.
    bottom_left_origin: (2,) array_like
        Location of the bottom left grid position in world coordinates.
    image_plane_distance: float
        Distance between pinhole origin and image, increase for more vertical cursor rays.
    grid_z: float
        The offset of the grid relative to the xy plane (so the z coordinate of the grid location).
    """

    # Create the origin location of the map, at the bottom left corner.
    rr.log(
        f"{prefix}",
        rr.Transform3D(
            translation=[bottom_left_origin[0], bottom_left_origin[1], grid_z]
        ),
    )
    height, width = grid.shape[0], grid.shape[1]

    # Create the origin of the pinhole transform at the center of the image, image_plane_distance below xy plane.
    # Rotate over x axis ensures up/down in the top down projection gets image bottom left to coordinate bottom left.
    pinhole_origin_x = width * resolution / 2.0
    pinhole_origin_y = height * resolution / 2.0
    pinhole_transform = rr.Transform3D(
        translation=[pinhole_origin_x, pinhole_origin_y, image_plane_distance],
        rotation=rr.datatypes.RotationAxisAngle(
            axis=[1, 0, 0], angle=rr.datatypes.Angle(rad=np.pi)
        ),
    )
    pinhole_projection = rr.Pinhole(
        focal_length=image_plane_distance / (resolution),
        width=width,
        height=height,
        image_plane_distance=image_plane_distance,
    )
    # Finally, log the transform, pinhole projection and the grid.
    rr.log(f"{prefix}/grid", pinhole_transform, pinhole_projection, rr.Image(grid))

```

In the viewer (with the ui lines set to zero, interactivity disabled in the top left panel):

[image]

This shows some debug markers I placed to verify the origin and scaling of the map. Full code to reproduce this scene; [log_grid_xy.py](https://github.com/user-attachments/files/23144061/log_grid_xy.py), with this manually created greyscale image for the logo;

[image]