rerun-io/rerun #1533: Support for "Derived" Transforms

jleibs, 2023-03-08:
## Issue Overview
In the current design, the transform graph must perfectly matches the entity path hierarchy.

However, there are situations where the user-data graph transform representation does not match the path hierarchy we want to use for logging data. For example, a URDF file may contain several linkages that logically exist to match the mechanical design of the robot, but are otherwise not interesting to the user.

At present the user must choose between two sub-optimal options:
 - Logging their data to paths that exactly match the transform graph allows them to use the transform values directly, but can lead to overly verbose path hierarchies that restrict freedom of users to organize data logically.
 - On the other hand, deriving path-relative transforms at log-time, as we do in the ROS example, allows the user to choose their own data organization, but sacrifices visibility into the original raw transform values.

## Proposal: "Derived" transforms.

The idea is that the value of the transform for that link in the path-hierarchy, rather than being directly fetched from the store, would be derived from some other function at render time.  Specifically, this must be a function of *time* (along with some other data logged to the store). However, the high-level transform semantics are crucially unchanged:  the transform (regardless of how its resolved) still represents the transform between those two spaces defined by the path hierarchy and is considered fixed for a given instant in time. As such, cross-space projection can be totally independent of of the resolver -- it just needs to know that at a given timepoint on a given timeline, the transform resolved to a particular value. Additionally the resolver need not play a role in space-partitioning and view-construction perspective. The mere fact that a "resolved transform" has been logged to a path allows us to do all of the same path-hierarchy analysis we already perform.

The current transforms already fit into this model. They simply use a "default resolver" that returns the latest-at value of the transform logged to that path.

There are two immediate mechanisms of resolved transforms that come to mind as being useful...

### "graph-lookup" transform

First, raw transforms would be logged with a graph-centric representation.

For example, adopting from a ROS tf system, transforms would be logged to a path derived from their child_frame_id: `tf/<child_frame_id>` and include a link to their parent_frame_id.
```
rr.log_rigid3_graph_link('tf/camera_optical_rgb', parent=''tf/camera_link", parent_from_child=...)
rr.log_rigid3_graph_link('tf/camera_link', parent=''tf/left_wrist_link", parent_from_child=...)
rr.log_rigid3_graph_link('tf/left_wrist_link', parent=''tf/left_arm_link", parent_from_child=...)
rr.log_rigid3_graph_link('tf/left_arm_link', parent=''tf/base_link", parent_from_child=...)
```

Then, a given path can be logged as a resolved "graph lookup" transform:
```
rr.log_rigid3_graph_lookup('robot/camera', parent='tf/base_link', child='tf/camera_optical_rgb')
```

At resolution time, this would then do a graph walk based on the logged topology.

### Parametric transform

Much simpler than graph-based resolution is simple support for parametric transforms.

Once we have resolved transforms, we can derive transforms from things like scalar values representing joint rotations or linear actuator offsets.

Exa:
```
rr.log_scalar('joint/arm', arm_rotation_rad)
rr.log_rigid3d_parametric_rotation("robot/arm", axis=(1,0,0), value="joint/arm")
```

This avoids the need for something like the "robot state publisher" in ROS.

nikolausWest, 2024-01-17:
The parametric transform could potentially be done using some kind of entity reference.
```python
rr.log('joint/arm', rr.components.Angle(radians=arm_rotation_rad))
rr.log("robot/arm", rr.Transform3D(axis=(1,0,0), angle=rr.components.Angle.Reference('joint/arm')))
```

Wumpf, 2025-06-02:
@methylDragon I think so, yes. The connectivity information should definitely be independent of the entity hierarchy and be a piece of data that can be relogged. Good you bring it up though, the viewer is today sometimes struggling with topology changes and we need to make sure we allow everything that doesn't completely mess with 2d/3d spaces (i.e. I think the viewer can be kept simpler by assuming that something that's under a pinhole stays under a pinhole and vice versa)

nikolausWest, 2025-12-02:
I think the thing I was talking about here was rather configuring the rotation angle once and then logging the angle many times. Will that be possible with the new setup?

Wumpf, 2025-12-02:
That is unfortuantely not yet possible and not part of any short term plans. I missed that the original ticket description talks about both graph transforms and parametric transforms. I thought there was a separate issue for that, but looks like I mixed things up!