rerun-io/rerun #1573: Rerun not working with Integrated Intel graphic card

chisarie, 2023-03-12:
Hi, first of all let me say that this library looks really promising, thank you a lot for the effort! I just stumbled upon it and I am looking forward to try it out more.

**The Issue**
When I run my script on the Desktop PC at University (which has an Nvidia RTX2080), everything works. When I run it on my laptop at home (which only has integrated Intel graphics), I have issues with the Graphic Drivers. 

**Screenshots**
[image]

Here the same copy-pasted:

```
2023-03-12T19:23:20.101027Z  INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 1    
2023-03-12T19:23:20.115877Z  WARN wgpu_hal::vulkan::instance: create_instance: ERROR_INCOMPATIBLE_DRIVER    
2023-03-12T19:23:20.118464Z ERROR eframe::native::run: Exiting because of error: Wgpu(DeviceError(RequestDeviceError)) on event Resumed
Traceback (most recent call last):
  File "/home/eugenio/miniconda3/envs/centergrasp_env/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/eugenio/miniconda3/envs/centergrasp_env/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/eugenio/miniconda3/envs/centergrasp_env/lib/python3.9/site-packages/rerun_sdk/rerun/__main__.py", line 15, in <module>
    main()
  File "/home/eugenio/miniconda3/envs/centergrasp_env/lib/python3.9/site-packages/rerun_sdk/rerun/__main__.py", line 11, in main
    exit(bindings.main(sys.argv))
RuntimeError: WGPU error: DeviceError(RequestDeviceError) -> DeviceError(RequestDeviceError)
```

**System Information**
 - OS: Ubuntu 20.04.5 LTS
 - CPU:  Intel® Core™ i7-10510U CPU @ 1.80GHz × 8 
 - Graphics: Mesa Intel® UHD Graphics (CML GT2)

Also to note that I believe Vulkan works properly on my system:
- `vkcube` runs well
- `vulkaninfo` output seems reasonable

**Question**

So the final question: is this library only suited for systems with discrete graphic cards? 
- If yes, it would be better to add a check at startup, with a clear error message to the user in case no discrete card is found. And also a mention in the docs.
- If no, then either I have something wrong in my drivers, or there is a bug somewhere, either in WGPU or in rerun I guess

jleibs, 2023-03-12:
Thanks for the report. We do expect things to work with intel graphics -- I can confirm I've been running with intel graphics successfully on Ubuntu 22.10, so this is more likely just a driver and/or config issue.

Our support for Ubuntu 20.04 is relatively new and not as thoroughly tested as the other platforms.

Can you see if any of the suggestions in our troubleshooting guide help? https://www.rerun.io/docs/getting-started/troubleshooting#graphics-issues

chisarie, 2023-03-12:
Hi, thank you for the super fast response. Yeah I tried to look into the troubleshooting guide, no luck unfortunately.

emilk, 2023-04-03:
Let's open an issue on the `wgpu` repo and see if anyone has any ideas

emilk, 2023-04-17:
@chisarie In order to diagnose this, it would be great if you could try out the official `wgpu` examples:

https://github.com/gfx-rs/wgpu/wiki/Running-the-examples

If they fail, please open an issue on https://github.com/gfx-rs/wgpu/ and link to it from here!

If it works, then perhaps the problem is in eframe. Please try this:

```
git clone git@github.com:emilk/egui.git
cd egui
RUST_LOG=debug cargo run -p egui_demo_app --no-default-features --features wgpu
```

Again, if there is a problem, open an issue on https://github.com/emilk/egui and link it here.

chisarie, 2023-04-17:
Hi Emil, 

When I opened the issue, the last release of rerun was 0.3.1. Just to be sure, I tried to run the `rerun_demo` script again today, and it was still failing. 

I was about to try the `wgpu` examples, but then I noticed that a new version of rerun came out in the meantime (0.4.0). So I installed the new version, and now the example works! 

So I guess the issue is solved, something has been changed in the last release that removed the problem. Thank you again for looking into this, I am definitely going to use rerun from now on for my visualization needs :)

closed 2023-04-17 by chisarie

Wumpf, 2023-04-17:
Huh interesting. We did a wgpu update in there which probably fixed it. But didn't expect that 😅 
Thanks for following up on the ticket! Looking forward to hearing again from you @chisarie now that you can actually use the tool 😄