rerun-io/rerun #1588: Min-max range for Turbo colormap doesn't make sense for depth images

nikolausWest, 2023-03-14:


emilk, 2023-03-23:
I fail to reproduce.

This is nyud, with the points invisible, and `Backproject Depth` turned on:

[image]

The turbo colormap works well, and the 2D view has no colormap, since we haven't gotten to that yet:
* https://github.com/rerun-io/rerun/issues/1612

nikolausWest, 2023-03-23:
@emilk 

I'll get an easier repro for you asap but in the mean time, has something changed so we don't use turbo by default for depth images? We set turbo as default for depth images here:

- https://github.com/rerun-io/rerun/issues/351

emilk, 2023-03-23:
We seem to only use it onlyfor float depth images, not integer depth images. I can make a fix for that.

nikolausWest, 2023-03-23:
For repro, just try:
- https://github.com/rerun-io/rerun/pull/1538

emilk, 2023-03-23:
`arkitscenes` have somehow been fixed by my recent PRs:

[image]

I'll merge `main` into the branch

nikolausWest, 2023-03-23:
> arkitscenes have somehow been fixed by my recent PRs:

Fantastic! Best kind of fix =)