rerun-io/rerun #1599: Need a simple example to use every primitive

asuper0, 2023-03-16:
**Describe the annoyance**

The existing examples are too complicated, core function of rerun are flooded with codes for data processing or algorithm. It's hard for me to find out usage of the primitives, especially for the rust language, for the lack of rust example.

**Expected behavior**

A simple example with very simple data, which shows basic usage of every primitives, in both rust and python.

jleibs, 2023-03-16:
Agreed. A nicely organized standalone version of each API, linked from the primitives docs is something we’ve been talking about for a while. Will try to get to that soon.

In the mean time the api_demo is probably the closest to what you might be looking for: https://github.com/rerun-io/rerun/blob/latest/examples/rust/api_demo/src/main.rs

Very little algorithm code in there and most of the sub-samples are focused on just one API surface. But not everything is covered or organized perfectly.

asuper0, 2023-03-17:
That could helps, thank you.
Also expect the one you metioned😁

emilk, 2023-03-20:
Related to [#1483](https://github.com/rerun-io/rerun/issues/1483)

emilk, 2023-04-17:
Let's expand on `api_demo` to include all primitives. Help wanted 🙏

jleibs, 2023-04-17:
Now that we have notebook support, I think we should take a stab at using it to document the python APIs.  We can maintain the doc- notebooks in the repository, and run them headless in CI using nbconvert. This both tests the APIs work and generates artifacts we can host.

For example
```
jupyter nbconvert --to=html --ExecutePreprocessor.enabled=True examples/python/notebook/cube.ipynb
```

Creates a standlone html document:
[image]

We can also push these same notebooks to colab and link as appropriate so that users can quickly play around with the API samples live if they want to experiment.

Wumpf, 2023-04-18:
Agreed, this would be a huge boost for our documentation! One of the best parts of this is that it makes it so much easier for us to write literate-programming-like examples, making the doc maintenance itself easier on us code-focused folks

emilk, 2023-04-18:
This is more user-facing, while we should also have a separate `api_test` that is completely exhaustive with all different tensor types, etc.

closed 2023-06-12 by emilk