rerun-io/rerun #1611: Varying opacity for spatial primitives

roym899, 2023-03-18:
**Is your feature request related to a problem? Please describe.**
Varying opacity can be useful to visualize overlapping or occluded data.

Seems like most primitives support RGBA arguments, however the alpha channel was ignored for the one I tested (`rr.log_line_segments`) or didn't correspond to opacity (`rr.log_mesh`).

**Describe the solution you'd like**
Would be great if all primitives (including 3D primitives) support some form of transparency / alpha compositing.

Wumpf, 2023-03-19:
Partially blocked by #702 - in planning this week we decided to get in some more transparency without order independence. I.e. we'll move forward to supporting it, but visual bugs will be expected when several transparent surfaces overlap on screen.

shabbirmarzban, 2023-10-05:
Hi there, what is the current status (of this feature)? I particularly would like to use it for meshes, thanks.

Wumpf, 2023-10-07:
@shabbirmarzban no progress and no eta so far sadly. But curious to hear what you're looking for exactly - e.g. if you'd get transparency for meshes how do you want to control it? (uniformly/per-vertex/via-texture, data via custom mesh or mesh format or set via viewer etc.)

julianschoep, 2024-01-25:
From a user perspective, for me the most straight-forward would be to have an opacity slider next to the entity visibility toggle, so entity-level control in the UI. Even if it's not ordered correctly it will add a lot of value for us.

Ipuch, 2024-04-04:
I have interest for this feature too

nshankar, 2024-04-08:
I also have interest in transparency. Are there any examples of how to correctly render 3D points with RGBA values? Here is a minimal example where the transparent point is rendered as opaque

```
import rerun as rr
from numpy.random import default_rng

rr.init("rerun_example_points3d_random", spawn=True)
rng = default_rng(12345)

positions = rng.uniform(-5, 5, size=[10, 3])
colors = rng.uniform(0, 255, size=[10, 3])
radii = rng.uniform(0, 1, size=[10])

rr.log(
    "transparent",
    rr.Points3D(positions=[(0, 0, 0)], colors=(0.5, 0.5, 0.5, 0.1), radii=1),
)
rr.log("random", rr.Points3D(positions, colors=colors, radii=radii))
```

Wumpf, 2024-04-09:
@nshankar this issue is about transparency not being available yet, i.e. you can't achieve this yet api code

georgematheos, 2024-05-17:
Hi, I just wanted to upvote this again.

In an application my team is working on, the ability to visualize meshes with different opacities is quite important.  (It is important enough that we are thinking of introducing meshcat visualizations in a codebase otherwise using the rerun visualizer, so we can produce visualizations like this.)  Having support for making such visualizations in rerun would be great!

In case helpful, here are a few more details about the desiderata we are encountering -- 
- A sufficient MVP for our application is if each mesh could be marked with an alpha value, to set the opacity of the whole mesh.
- If it were possible to have each triangle in the mesh have its own independent opacity, that would be even better. 
- It is necessary for our application that the opacity values can be set programmatically (as the alpha values we want in the visualizations are computed via numerical calculations).

Thanks!

dojoscan, 2024-10-08:
> Hi, I just wanted to upvote this again.
>
> In an application my team is working on, the ability to visualize meshes with different opacities is quite important. (It is important enough that we are [...]

My team have an application with the very similar requirements. Would be great to have this feature!

leonmkim, 2024-12-03:
Just want to echo everyone above in their interest in this feature. I work in robotics and would like to visualize planned trajectories. Using a sequence of coordinate axes is ok, but a trajectory of object meshes/primitives with decaying opacity through the trajectory would be much more useful and is a common approach. A uniform opacity per mesh would be totally sufficient. For an example, see [video](https://dynamic-controlled-sliding.github.io/videos/hwlog35_w_legend.mp4) courtesy of [Will Yang](https://dynamic-controlled-sliding.github.io/).

Any partial support would be great (e.g. only support native rerun primitives as a first step). Lack of opacity support is a bit of a dealbreaker for many CV/robotics applications.

JAEarly, 2025-06-15:
Is there any progress on this? Support for alpha values in `LineStrips3D` and `Mesh3D` would be great for a project I'm working on :)

arijit-dasgupta, 2025-08-07:
Hi! Was wondering if this was in the radar, it would be extremely useful to implement transparency of these spatial primitives. I am also aware of several other rerun users who are frustrated by the lack of this feature. Thank you!

aedm, 2025-09-24:
Rerun 0.25 added basic transparency support.
Please share your experience. :)