rerun-io/rerun #1645: Visible history does not respect times correctly

roym899, 2023-03-21:
**Describe the bug**

Not sure what is the exact logic right now, but at least setting a history in seconds yields the wrong result right now.

**To Reproduce**
```python
import numpy as np
import rerun as rr
rr.spawn()
rr.set_time_seconds("test", 0)
rr.log_points("point", np.random.rand(5,3))
rr.set_time_seconds("test", 10)
rr.log_points("point", np.random.rand(5,3))
```
Moving to the last message, and setting the history to a number >0 will show all ten points. 

**Expected behavior**
Should only show 5 points, until history >=10.

**Screenshots**
[image]
Note that the data is 10s apart on the timeline, yet it shows both with a history of 0.1s.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

jleibs, 2023-03-21:
The behavior you're observing is "technically correct" but it's a bit nuanced and I see why it's unexpected without better documentation on the rerun temporal model of persistence. Going to leave this open until we can at least make this more clear.

The Rerun model is generally that "All logged components exist until they are replaced."  Although it's not indicated on the timeline as such, you can think of this as meaning that the first set of 5 points is "valid" over the semi-open interval [0, 10) and the second set of 5 points is valid over the semi-open interval [10, Inf).

As such, the visible history query on the interval: [9.9, 10] intersects with *both* of these validity intervals and subsequently both versions of the points are displayed by the query.

roym899, 2023-05-22:
Makes sense to me. I'll close this for now.