rerun-io/rerun #1681: Whenever the browser looses focus, camera rotates on click+drag

Wumpf, 2023-03-23:
It seems browser-based rerun thinks the Alt modifier key is pressed until one presses & releases it. This causes camera control to rotate on drag whenever the browser looses focus. Once I press `alt` it behaves as it should again.

May need to revert  #1659 if we don't find a proper fix

Observed so far on Firefox @ Windows.

emilk, 2023-03-23:
This sort of "sticky-modifiers" bug is very common. Please check if you have the issue on egui.rs:

[image]

Wumpf, 2023-03-27:
confirmed that it is a sticky modifier problem

closed 2023-03-27 by emilk

emilk, 2023-03-31:
Fixed upstream in https://github.com/emilk/egui/pull/2857