rerun-io/rerun #1704: Restructure our CI workflows

jleibs, 2023-03-23:
## Context
Our github CI workflows have grown organically over the last 6 months and are becoming somewhat haphazard / fragile. We've learned a number of tricks through that time that are helpful, but not everything is applied consistently.

### Major pain points:
 - Python code is not being tested on PRs.
 - Release jobs aren't predictably executed. (Seems like an inconsistency between the *triggering* vs the *filtering* of tag-related changes).
 - Need to build all wheels to create new rrd files even though only linux is necessary.
 - Debugging & testing changes to the workflows themselves are slow.

## Proposal
Here is an alternative workflow arrangement refactoring things into 4 major workflows.

### 1. Lint, Build, Test
This is the primary workflow that contains most of the current jobs. By default runs on linux only, but windows / mac can be included when running via workflow_dispatch input..

#### Triggers:
 - any commit on a PR
 - any commit pushed to main
 - manually via workflow input flags for which OS to build/test.
 - nightly build across all OSes

#### Jobs:
- Linting for both rust and python.
- Matrixed:
  - Build & test rust code.
  - Build python wheel
  - Run python tests using built wheel.
- Benchmarks (optionally via dispatch input or push to main).
 
### 2. PR Gates
This is a small workflow that contains all jobs related to meta things like required labels and tags, keywords that should prevent merging, too-large-of-files, existence of invalid file patterns (symlinks?) etc.

#### Triggers:
 - any commit on a PR

### Create Packages
This workflow is *just* for preparing cargo packages, wheels, rrd files, and app.rerun.io wasm bundles.  All artifacts are pushed to gcloud buckets based on the commit hash. This does not deploy to crates.io or pypi.

Almost of the things that could be built should be parameterizable via the job input.

#### Triggers:
 - manually
 - periodic "prerelease" (maybe hourly) if there are changes on head of main.
 
#### Jobs (all optional):
 - Build linux wheel
 - Build wasm bundle
 - Create RRD files using linux wheel
 - Build additional wheels
 - Create python documentation
 - Create rust documentation
 - Upload Python docs to gcloud
 - Upload Rust docs to gcloud
 - Upload RRD files to gcloud
 - Upload wasm bundle to gcloud
 - Create tag / draft release.
 
### Release
This is another small workflow that just orchestrates uploading of pre-built artifacts from gcloud to crates.io, pypi. The script should be doing no runtime building -- only download/validation/upload/signing etc.

#### Triggers:
 - manually only

#### Jobs:
 - Publish pypi packages
 - Publish crates.io packages
 - Update 'latest' and 'prerelease' tags and/or gcloud symlinks
 - Mark release as published

TODO:
 - [ ] Draw the Proposed Execution Graph

Wumpf, 2023-03-24:
Sounds really good to me overall, but I'm worried about having the PR check that small. It sure would be cheaper but that will lead to a lot more red main branch

jleibs, 2023-03-24:
Both 1 and 2 should block PR. I need a better name for 1 though to make that more clear. 1 is about the meta content of the PR and 2 is about the code in the PR.

emilk, 2023-04-03:
Maybe related:
* [#1754](https://github.com/rerun-io/rerun/issues/1754)

emilk, 2023-04-19:
Another CI fail: `just py-format` changes the `cube.ipynb`, but the CI didn't fail on it 🤦

closed 2023-04-27 by emilk