rerun-io/rerun #1714: Make timezone used to display time configurable

nikolausWest, 2023-03-27:
It's annoying to have to mentally convert between UTC time and my local timezone when looking at timestamps in the UI.

emilk, 2023-04-06:
The minimal version of this should provide a global option of showing times either as UTC, or in the local timezone.

This should be applied to all times in the UI, which means everywhere we call `time::format_description`, and most of the interface of `re_log_types::Time`

ParthJadhav, 2023-04-06:
Where would that option should be @emilk ?

emilk, 2023-04-17:
It should be in `AppOptions` in `crates/re_viewer/src/misc/app_options.rs`

(sorry for the late reply - I missed this one 😬)

jparismorgan, 2023-09-28:
Hey @nikolausWest / @emilk, took a pass at addressing this here: https://github.com/rerun-io/rerun/pull/3530 - and also had a few questions & comments on there as well!

closed 2023-10-03 by Wumpf