rerun-io/rerun #1738: `arrow2`'s `estimated_size_bytes` is _extremely_ slow

teh-cmc, 2023-03-30:
In fact it is so slow that it is pretty much impossible to use it anywhere in the store without everything slowing down to a crawl, which makes implementing statistics really, _really_ painful.

```rust
const NUM_ROWS: usize = 10_000;
const NUM_INSTANCES: usize = 100;

fn arrow_size_bytes(c: &mut Criterion) {
    let mut group = c.benchmark_group(format!(
        "arrow_size_bytes/rows={NUM_ROWS}/instances={NUM_INSTANCES}"
    ));
    group.throughput(criterion::Throughput::Elements(NUM_ROWS as _));

    {
        fn generate_cells() -> Vec<DataCell> {
            (0..NUM_ROWS)
                .map(|_| DataCell::from_native(build_some_point2d(NUM_INSTANCES).as_slice()))
                .collect()
        }

        let cells = generate_cells();
        let total_size_bytes = cells.iter().map(|cell| cell.size_bytes()).sum::<u64>();

        group.bench_function("cells", |b| {
            b.iter(|| {
                assert_eq!(
                    total_size_bytes,
                    cells.iter().map(|cell| cell.size_bytes()).sum::<u64>()
                );
            });
        });
    }

    {
        fn generate_vecs() -> Vec<Vec<Point2D>> {
            (0..NUM_ROWS)
                .map(|_| build_some_point2d(NUM_INSTANCES))
                .collect()
        }

        let vecs = generate_vecs();
        let total_size_bytes = vecs
            .iter()
            .map(|vec| std::mem::size_of_val(vec.as_slice()) as u64)
            .sum::<u64>();

        group.bench_function("vecs", |b| {
            b.iter(|| {
                assert_eq!(
                    total_size_bytes,
                    vecs.iter()
                        .map(|vec| std::mem::size_of_val(vec.as_slice()) as u64)
                        .sum::<u64>()
                );
            });
        });
    }
}
```

```
arrow_size_bytes/rows=10000/instances=100/cells
                        time:   [355.99 µs 357.00 µs 358.04 µs]
                        thrpt:  [27.930 Melem/s 28.011 Melem/s 28.091 Melem/s]

arrow_size_bytes/rows=10000/instances=100/vecs
                        time:   [4.2832 µs 4.2944 µs 4.3065 µs]
                        thrpt:  [2.3221 Gelem/s 2.3286 Gelem/s 2.3347 Gelem/s]
```

Even for incremental measurements this is way too slow, orders of magnitudes slower than everything else on the write path.

---

Assuming we cannot optimize it any further, my proposal is to move the problem upstream: compute byte sizes within the batching system, therefore:
1. distributing the load to the clients
2. making its cost irrelevant, since batching happens on a separate thread

emilk, 2023-03-31:
Can't we fix this in `arrow2`? It sounds like a performance bug.

teh-cmc, 2023-03-31:
> Can't we fix this in `arrow2`? It sounds like a performance bug.

Yep, I want to investigate this tomorrow after a good night's sleep.
Sounds like a good opportunity to become _really_ intimate with `arrow2`'s internals too, which I've been meaning to do for a while anyhow.

teh-cmc, 2023-04-01:
Long-winded explanation [here](https://github.com/rerun-io/rerun/pull/1743#issuecomment-1493077945), but the TL;DR is that, as far as I can tell, A) there is nothing wrong with how `estimated_bytes_size` is implemented and B) performance is getting absolutely wrecked by nested virtual calls.

Due to the way `arrow2` is structured, the number of virtual calls necessary to execute `Array::clone` or `estimated_bytes_size`  scales linearly with the number of fields in a type, while at the same time the performance of these methods decreases pretty much linearly with the number of virtual calls involved.
I.e. cloning or estimating the size of a `Point3D` will be much slower than doing the same thing on an `InstanceKey`.

At least in the short-term I think the solution stays the same: get these operations out of the fast path, cache the results in `DataCell`s, use `DataCell`s everywhere, avoid working with raw arrow arrays as much as possible.

closed 2023-04-12 by teh-cmc