rerun-io/rerun #1741: Allow hiding back/front faces of meshes

Wumpf, 2023-03-31:
All meshes are rendered double sided currently since we don't know what is front/back (it's hard to enforce a standard!). This is a resonable default for Rerun, but there should be a blueprint option for backface culling where one can specificy whether clockwise/counterclockwise triangles are shown. Need to figure out good names for this.

emilk, 2023-04-03:
Will help the ARKit example

Wumpf, 2023-04-03:
Bunch of things to do for the proper solution:

* separate semantics of mapping front/back <-> CW/CCW and showing All/front/back
* read this mapping from mesh format if defined (many formats do!)
* establish what our default mapping is
* APIs for all of the above

emilk, 2023-04-04:
This will blow up the required number of mesh pipelines by x3 if I'm not mistaken

emilk, 2023-04-11:
Related: https://github.com/rerun-io/rerun/issues/701

emilk, 2023-04-18:
This is both a component, and should be overridable in the blueprint store