rerun-io/rerun #1748: `smallvec` performance issues on common store operations

teh-cmc, 2023-03-31:
See #1747 for detailed benchmarks.

The obvious solution seems to be to simply switch to `tinyvec` which:
- doesn't exhibit the same issues
- doesn't have any unsafe code
- has more features

The only downside is the `Default` requirement: for those cases where `Default` is not an option, a vanilla `Vec` seems like the best bet.

Wumpf, 2023-03-31:
> a vanilla Vec seems like the best bet.

not necessarily though I'd argue: Sure your benchmark shows quite a few issues on access performance, but I'd expect `SmallVec`s that stay in their "small bound" to still have way better behavior access behavior on cold cache (which is always a great claim to put out there because it's so hard to prove, heh) than `Vec`. I mean, all the reasons for why one would like to avoid the heap allocation/`Vec` are still there obviously

teh-cmc, 2023-04-04:
This is appropriately documented and benchmarked now so this issue effectively fixed; each system will have to act accordingly depending on their tradeoffs. Closing.

closed 2023-04-04 by teh-cmc