rerun-io/rerun #1790: Load data from the .rrd file

asuper0, 2023-04-06:
**Is your feature request related to a problem? Please describe.**

In some case, the data I transferred to rerun is not only a final result, but also a intermediate result. Which means there would be further process with these data. For example, some fft result. You may log them with rerun and get some key information by the fft graph, and then deal it with your algorithm code.

**Describe the solution you'd like**

It would be great to load raw data from a .rrd file with rerun interface, rather than save and load datas in another way.

emilk, 2023-04-06:
Do you mean that you would like to use our logging SDKs to load and .rrd file and stream it to the viewer?

asuper0, 2023-04-06:
No, I want to load data from .rrd and do some other calculation. As metioned, sometimes I will gather realtime data from the device, a radar for example, and want to see fft curve in realtime. If I'am satisfied with the data, I would save them to .rrd file, and then use an other program to deal with the data.

asuper0, 2023-04-06:
Some time I would decide to discard the data just by looking at the FFT result. Some time I would modify my code depend on the FFT result. 
When I decide to keep the data, I will use them as input to other algorithm. That's why I call it a intermediate result.

nikolausWest, 2023-04-10:
I recognize this use case from my own personal experience and we plan to support this in the future. To start with it will probably be through the SDK where you will be able to run queries on recordings (basically `.rrd` files) that return dataframes. There are a couple other pieces that need to land first though

vmayoral, 2024-07-10:
Now that 0.17 is out, any ETA in here folks?

teh-cmc, 2024-07-11:
> Now that 0.17 is out, any ETA in here folks?

Building the infrastructure that allows users to query their data back is one of our focus point for this summer.
My conservative ETA for when this will make its way into an actual release would be some time during autumn.

vmayoral, 2024-08-18:
Hello again, thanks for the release of `0.18.0`, great improvements.

Concerning this feature, I saw in your [roadmap](https://rerun.io/docs/reference/roadmap) that `Query API for reading back data from the SDK` is expected in Q3, which I believe will address this issue. Can you confirm if this assumption is correct?

Thanks!

nikolausWest, 2024-08-22:
We are currently working on that feature. It may slip into the beginning of Q4 but it's our highest priority right now

teh-cmc, 2024-10-19:
https://rerun.io/docs/getting-started/data-out

closed 2024-10-19 by teh-cmc