rerun-io/rerun #1816: Rerun packages available on conda

jleibs, 2023-04-11:
There's already a good starting point to build off of: https://github.com/conda-forge/staged-recipes/pull/22471

TODO:
 - [ ] Finish the recipes for Mac & Windows support.
 - [ ] Add to our release flow

emilk, 2023-04-18:
How will this be maintained?

ruben-arts, 2023-05-16:
Hey, I could give you some extra info. 

The staged-recipes result in a feedstock, which automates build setup regeneration based on GitHub releases completely in their CI. 

The testing part can be extended with CI tests and/or you can make the merge of the generated update manual. Then the package can be tested locally first. 

In my experience, it can require some time to do the initial setup but updates are straightforward afterward, it's a simple push of a button. Plus there is a helpful community where issues get resolved easily. 

If you would be afraid of ever pushing bad code/product there is a system in place to deal with that check: [Conda-forge Repodata Patches](https://github.com/conda-forge/conda-forge-repodata-patches-feedstock)
This system avoids installation on systems without the sdk in a lockfile, while maintaining lockfile functionality.

ruben-arts, 2023-06-30:
Hey, we just worked on fixing the feedstock pull request. 

It now works on `windows` `osx-64` and `linux-64`. `osx-arm64` can be done after the first build. 

Currently I'm the only maintainer but it would be nice if one of you joined. 

https://github.com/conda-forge/staged-recipes/pull/22471

closed 2023-08-04 by nikolausWest