rerun-io/rerun #1843: Ctrl-C still hangs when hosting web-server in some cases

jleibs, 2023-04-14:
Not sure which OS/browser combinations this hits.  For me if I launch with `rerun --web-viewer`, and then ctrl-c it doesn't exit until I refresh the chrome tab that's opened.

Looks like we're hitting some variant of:
 - https://github.com/hyperium/hyper/issues/1885

Wumpf, 2023-04-19:
reproed on Windows. Can't shutdown from `python .\examples\python\arkitscenes\main.py --serve`, refresh does not fix it for me

Wumpf, 2023-04-20:
Above report for windows --serve is fixed in https://github.com/rerun-io/rerun/pull/1939 but this has nothing to do with with the way Jeremy reports starting the viewer which isn't from python

zehiko, 2024-12-11:
I've just hit this on Mac. I ran a simple python example:

```
#!/usr/bin/env python3

import rerun as rr

rr.init("stuck")
rr.log("entity",rr.Points3D([[0,0,0],[1,1,1]], radii=[0.5,0.5,0.5]))
rr.serve_web()
```

Ctrl-C didn't work, needed to kill -9 the process.

jleibs, 2024-12-11:
The behavior using the script from @zehiko above appears to have regressed between 0.16 and 0.17, however a more common pattern like:
```python
#!/usr/bin/env python3

import rerun as rr
import time

rr.init("stuck")

rr.serve()

while True:
    rr.log("entity", rr.Points3D([[0, 0, 0], [1, 1, 1]], radii=[0.5, 0.5, 0.5]))
    time.sleep(1)
```
still exits on Ctrl-C just fine.

jleibs, 2024-12-11:
Created a new issue for the above -- turns out this is less about Ctrl-C and more about a dead-lock when draining the backlog:
https://github.com/rerun-io/rerun/issues/8417