rerun-io/rerun #1845: Replace `anyhow` with `thiserror` in libraries

emilk, 2023-04-14:
We have been too liberal of our use of `anyhow`. For any `pub` function in our libraries, we should be using `thiserror` instead of `anyhow`, to get more type safety and explicitness of what can go wrong.

Internal use of `anyhow` in a crate is less bad, but would also be good to change to `thiserror`.

`anyhow` should mostly be used in high-level crates (binaries) to easily summarize errors coming from many different places without having to create too many error `enum`s

emilk, 2024-07-14:
We should do this in `re_types` in particular, and somehow forbid it so that it doesn't come back.

ishakbhatt, 2025-04-10:
I have replaced it in `re_types` and am doing a proper verification that it works before making a draft pull request. Do we want to include all of the other crates in a single pull request or would you prefer doing this with `re_types` and scoping it later to include other crates? @emilk

grtlr, 2026-03-24:
I feel like this issue is causing churn, so I plan to close it after:

* #12702 

has landed. Thanks to the contributions, we already got the usage of `anyhow` down quite a bit. In the future, these kind of refactors should probably be motivated by actual cases where the current error reporting is not enough.