rerun-io/rerun #1867: Using WASD+EQ with Ctrl Modifier results in program exit & save file.

h3mosphere, 2023-04-16:
It seems that WASDEQ speed can be modified via Ctrl|Shift|Alt. This is nice, however it appears to clash with application level keyboard shortcuts: Ctrl-Q, and Ctrl-S.

**To Reproduce**
Steps to reproduce the behavior:
1. Press Ctrl-Q or Ctrl-S, expecting fine movement in the Y-Up, or Z-Back direction.
2. The application quits, or pops up a save dialog.

**Expected behavior**
1. Maybe remove the application level key bindings. Needing to quit the application, and save, come up alot less frequently than fine-level control of camera, for certain data sets.
2. Have Shift + WASDEQ be the fine-level movement, instead of control. Perhaps even removing Control entirely. Currently Shift moves _very_ fast, for a dataset which is 70m+ in the X+Z directions, the dataset flies out of the view very easily.

**Your goals**
Move around the dataset to get a view from a particular point, needing quite fine level control over movement.

**Desktop (please complete the following information):**
 - OS: Linux Ubuntu 22.10

Wumpf, 2023-04-16:
Thanks for filing this!
(For all Mac users out there: You don't run into this problem because saving is via Cmd for you!)

imho neither removing the shortcuts nor removing the slow-down is great:
* application wide shortcuts at least for saving are so common, they are expected to work
* speed *plus* slowdown are useful to quickly support variety of scenes. But we may need to tweak the factors  & expose camera speed etc. as a user setting on the space views

With that stance the only way out is a different button for slowdown. Proposing to stick with Ctrl on Mac and use Alt on Windows & Linux. What do you think?

h3mosphere, 2023-04-17:
Whichever seems most appropriate. Thinking about it now, it is probably best to stay away from CTRL as it is used most often with key combos that users will be familiar with.

Alt may already be bound to a speed 'mode'. I'll have a look at the the code see what the combinations are currently. 

Probably for someone else to weight in here on what is correct/blessed.

Thanks

PR #3051 (2023-08-31): Change the "slow-down-camera" modifier to Alt on non-Mac
This Fixes #1867. Setting Alt as the slowdown behaviour for the camera movement. Note, that this also sets it to alt for wasm architecture, let me know if you wish for an extra check for wasm (and leave it as alt).

Thanks.

### What

### Checklist
* [X] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [X] I've included a screenshot or gif (if applicable)
* [X] I have tested [demo.rerun.io]( [... cut]
commit 990a6258 (2023-08-31): Change the "slow-down-camera" modifier to Alt on non-Mac (#3051)

This Fixes #1867. Setting Alt as the slowdown behaviour for the camera
movement. Note, that this also sets it to alt for wasm architecture, let
me know if you wish for an extra check for wasm (and leave it as alt).

Thanks.

### What

### Checklist
* [X] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [X] I've included a screenshot or gif (if applicable)
* [X] I have tested [demo.r [... cut]
commit dbe3c824 (2023-08-31): Change the "slow-down-camera" modifier to Alt on non-Mac (#3051)

This Fixes #1867. Setting Alt as the slowdown behaviour for the camera
movement. Note, that this also sets it to alt for wasm architecture, let
me know if you wish for an extra check for wasm (and leave it as alt).

Thanks.

* [X] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [X] I've included a screenshot or gif (if applicable)
* [X] I have tested [demo.rerun.io](https://demo.re [... cut]
commit 46c8d85d (2023-08-31): Change the "slow-down-camera" modifier to Alt on non-Mac (#3051)

This Fixes #1867. Setting Alt as the slowdown behaviour for the camera
movement. Note, that this also sets it to alt for wasm architecture, let
me know if you wish for an extra check for wasm (and leave it as alt).

Thanks.

* [X] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [X] I've included a screenshot or gif (if applicable)
* [X] I have tested [demo.rerun.io](https://demo.re [... cut]
closed 2023-08-31 by emilk