rerun-io/rerun #1871: Show tensors shaped `[H, W, 1, 1]` as images

emilk, 2023-04-17:
Mentioned [here](https://github.com/rerun-io/rerun/issues/1732#issuecomment-1505967290) and elsewhere:

It is fairly common to store a single image in a `[H, W, 1, 1]` or `[H, W, 3, 1]` tensor.

However, currently Rerun only shows `[H, W, 1|3|4]` tensors as images, despite the trailing unit-dimension beeing an noop.

This should be fairly easy to handle by ignoring trailing unit-dimensions in:
* `Tensor::image_height_width_channels`
* `Tensor::is_shaped_like_an_image`
* `Tensor::is_vector`

Preferably they would just use the same function, `fn shape_short(&self) -> &[TensorDimension]`

tauseefmohammed2, 2023-04-18:
Hi!
I would love to contribute to the repo. I saw that the issue is labelled "first time", which is perfect for me as I am new to this project and eager to get started. Could you please assign this issue to me?

If I am not wrong, this is a straightforward implementation change that can be made. The fucntion can simply check if the last dimension of the tensor is a unit-dimension, and if so, remove it before displaying the image.

Wumpf, 2023-04-18:
@tauseefmohammed2 sure go ahead! We actually rarely use the assign feature 😄 , but this way at least nobody else starts working on it redundantly :)
Yeah I think you're right should be as straight forward as you say - returning a subslice of the dimension that ignores trailing unit dimensions. Tbh not sure why @emilk  calls this _sort_ (:

commit 3601d2e1 (2023-05-10): Show tensors shaped [H, W, 1, 1] as images (and more!) (#2075)

* Ignore trailing tensor dimension when determining whether a tensor is an image.
Fixes #1871
Quite a bit of nuance to support single channel 1x1 images & line-like images.

* fix image preview for images other than M x N x C and M x N

* comment fix

* better shape_short comment

* handle empty tensors

* is_shaped_like_an_image is now defined via image_height_width_channels, improve comment on both

* any Nx1x... image now now treated as image

* rename to get_with_image_coords, make it more restrictive

* tensor_to_gpu height_w [... cut]
commit 45ebf13a (2023-05-11): Show tensors shaped [H, W, 1, 1] as images (and more!) (#2075)

* Ignore trailing tensor dimension when determining whether a tensor is an image.
Fixes #1871
Quite a bit of nuance to support single channel 1x1 images & line-like images.

* fix image preview for images other than M x N x C and M x N

* comment fix

* better shape_short comment

* handle empty tensors

* is_shaped_like_an_image is now defined via image_height_width_channels, improve comment on both

* any Nx1x... image now now treated as image

* rename to get_with_image_coords, make it more restrictive

* tensor_to_gpu height_w [... cut]