rerun-io/rerun #1873: Refactor `re_viewer`

emilk, 2023-04-17:
It is growing big and unwieldy. We should aim to break it up into more crates, and clean up some stuff.

Ideally `re_viewer` should only be UI code.

This needs some more planning before we execute, but some rough ideas:

* [ ] Move blueprinty non-gui stuff to own crate (re_blueprint)
* [ ] Break up `app.rs` (make it a module, and show multiple loaded recordings as tabs)
* [ ] Refactor the different views (view_tensor, view_text, …), and maybe move them to their own crates
* [ ] …

Wumpf, 2023-04-18:
Keep #1164 in mind while working on this

Wumpf, 2023-05-02:
Started internal writeup on possible splits https://www.notion.so/rerunio/re_viewer-splitup-refactor-8f0c96406e5a4d5e9d030e74d7a6db34

Wumpf, 2023-06-09:
landscape changed quite a bit by now. I'd call that done despite there being a bunch of things that could still be separated out (most prominently the Blueprint panel and the Selection panel) and App.rs could look better still

closed 2023-06-09 by Wumpf