rerun-io/rerun #1889: Minimal control of views through SDK

jleibs, 2023-04-18:
The long-term goal for controlling the UI is introduction of a blueprint store for most UI state: https://github.com/rerun-io/rerun/blob/main/design/blueprint_store.md

However, we can create immediate value for users (in particular in a notebook context) with just a subset of the functionality from the RFC.

In order of priority:
 - [ ] Specify panel state (collapsed or expanded)  for the 3 panels
 - [ ] Specify selected timeline
 - [ ] Specify whether to enable auto-view creation (this should be one of the first things logged; probably part of init?)
 - [ ] Specify views to be created (view name, space root, and list of entity paths)
 - [ ] Specify entity properties 

All other behavior can still be determined by heuristics.

There are a few open design questions that warrant further discussion:
 1. Do the UI msgs go over the same channel as the other data, as a single stream/rrd or is it a fully parallel pipeline?  Essentially, do you "save" a data rrd separately from saving a "ui rrd?"
 2. How do UI initialization via SDK interact with local persisted viewer modifications.  Exa: application logs UI events, then user makes local edits, then runs application again, which logs ui events. Is there a deterministic state we end up in?

Proposal 1:
 - As a starting point, default the behavior to on the first "UI" message always reset the state. If the user has local edits, maybe: present a modal asking the user to choose whether to throw away their current state or ignore the view events.  

Proposal 2:
 - Keep 2 copies of the blueprint store (maybe just 2 separate timelines?).  One store is only the message received from the app, so it will always be a deterministic state, while the other also includes the user edits. Introduce options to toggle which mode is active, and to reset the user-ui back to the app-controlled state (or to the auto state).

emilk, 2023-04-18:
Let's get started on the design of this ASAP

jleibs, 2024-03-15:
Done in:
 - https://github.com/rerun-io/rerun/pull/5408
 - https://github.com/rerun-io/rerun/pull/5484
 - https://github.com/rerun-io/rerun/pull/5498

closed 2024-03-15 by jleibs