rerun-io/rerun #1893: Fix splats all around (rs sdk, py sdk, re_query...)

teh-cmc, 2023-04-18:


Wumpf, 2023-04-18:
Created separate ticket for that ⬆️

jleibs, 2023-04-18:
> within a row with num_instances == 1, a component of unit length is not a splat

I've mentioned it elsewhere, but I want to note that the legacy explicit-splat design did allow us to differentiate between these two calls:
```
rr.log_points("points", positions = [p1], colors = csplat)
rr.log_points("points", positions = [p1], colors = [c1])
```
which have different behavior if later followed by the call:
```
# Just log new positions; rely on previous splat for colors.
rr.log_points("points", positions = [p1, p2])
```

We can work around this by, again, logging a separate row, but then we're back to having multiple rows again. Alternatively we can *only* log the extra row if the primary component is length-1 but that special-casing feels worse to me because it's going to create a host of new special edge-case handling for the 1-element path, which starts to feel a lot like going back to having mono-records.

teh-cmc, 2023-10-10:
- Related: https://github.com/rerun-io/rerun/pull/3730#discussion_r1352266630
> That's where it becomes funky...
> 
> If you do this:
> ```python
> rr.log("random", rr.Points3D(positions, colors=colors, radii=radii))
> rr.log_components("random", [rr.components.ColorBatch([255, 0, 0])])
> ```
> 
> Then you're going to end up with the original `colors` being discarded, a single red point and the rest of the points using the default color for this entity path (because that `ColorBatch` is _not_ a splat).
> 
> Now, there is a trick at your disposal... you could do this:
> ```python
> rr.log("random", rr.Points3D(positions, colors=colors, radii=radii))
> rr.log_components("random", [rr.components.ColorBatch([255, 0, 0])], num_instances=2)
> ```
> 
> And now you'll end up with only red points, because you explicitly said that the data was 2 instances wide, and so the log function considers the `ColorBatch` to be a splat...
> 
> Of course we could change things so that logging 1 thing is always considered a splat, but then you have the opposite problem, which might or might not be better depending on the situation :shrug:.
> 
> [And this is why I don't like that splats are a logging-time rather than a query-time concern](https://github.com/rerun-io/rerun/issues/1893): the view should get to decide what to do with the data that it has as its disposal, and that behavior should be configurable through blueprints and through the UI.
> This instance key business is pretty similar to e.g. configurable texture clamping modes in gfx APIs after all.

teh-cmc, 2023-10-12:
Here's another interesting case:
```python
import rerun as rr
import numpy as np

rr.init("rerun_example_arrows_splat", spawn=True)

vertex_positions = np.array([[-1.0, 0.0, 0.0], [1.0, 0.0, 0.0], [0.0, 1.0, 0.0]], dtype=np.float32)

rr.log(
    "triangle",
    rr.Mesh3D(
        vertex_positions=vertex_positions,
        vertex_normals=[0.0, 0.0, 1.0],
        vertex_colors=[[255, 0, 0], [0, 255, 0], [0, 0, 255]],
    ),
    [rr.Arrows3D.indicator()],
)
```
[image]
[image]

Look at how the Arrow is rendered.

The arrow was logged as a splat, so it has a single entry whose instance-key is `u64::MAX`.

Since we also logged an arrow indicator, the viewer will query an Arrow3D archetype using that same mesh data, except this time the arrow is our primary/required component, not an optional one.
The archetype query will try to join our one single arrow with the available positions and colors, but since position and color are not splatted, there isn't a single instance key in common and so the join doesn't yield anything.
The result is that we draw the arrow on its own, with the default origin and default color.

The "fix" is to log the normal multiple times so that it doesn't get marked as a splat at log-time:
```diff
-        vertex_normals=[0.0, 0.0, 1.0],
+        vertex_normals=[[0.0, 0.0, 1.0], [0.0, 0.0, 1.0], [0.0, 0.0, 1.0]],
```
[image]

This would not be an issue if splatting was a query-time concern.

---

Another somewhat related issue here is how our query model works in and of itself: since we join optional components onto primary components, even if the above issue was fixed, we would still only show a single arrow (properly joined with the first position and first color).

PR #6104 (2024-04-26): Remove instance keys and explicit splatting everywhere
Annihilate everything related to instance keys.

---

- [x] Remove `InstanceKey`
- [x] Remove `NumInstances`
- [x] Remove explicit/client-side splats
- [x] Remove cluster keys and autogenerated cells
- [x] Clean up every other little thing related to any of the above
- [x] Update batching docs
- [x] Check that Kiss-ICP now works smoothly
- [x] Check roundtrips
- [x] Check `rerun snippets/*.rrd`
- [x] Rebase and run checklist
- [x] Run full check bot

---

Findings (none blocking, not sure any are new -- to be investigated later):
- Depth c [... cut]
commit 893de2f4 (2024-04-26): Remove instance keys and explicit splatting everywhere (#6104)

Annihilate everything related to instance keys.

---

- [x] Remove `InstanceKey`
- [x] Remove `NumInstances`
- [x] Remove explicit/client-side splats
- [x] Remove cluster keys and autogenerated cells
- [x] Clean up every other little thing related to any of the above
- [x] Update batching docs
- [x] Check that Kiss-ICP now works smoothly
- [x] Check roundtrips
- [x] Check `rerun snippets/*.rrd`
- [x] Rebase and run checklist
- [x] Run full check bot

---

Findings (none blocking, not sure any are new -- to be investigated
later):
 [... cut]
closed 2024-04-26 by teh-cmc