rerun-io/rerun #1904: Garbage collection should be aware of `app_id`/`recording_id` semantics

teh-cmc, 2023-04-18:
We've seen plenty of reports of users that start a Rerun instance and then run their algorithm a bunch of times as they go through their iterative improvement cycle.

It ends up looking like a little this:
[image]

Now obviously at some point these users run out of memory, at which point they learn about `--memory-limit`.  
That's all fine except that garbage collection is completely unaware of these `app_id`/`recording_id` semantics, and so will only purge the currently active datastore (which likely contains the only data that the user still cares about at this point) while the old recordings are left untouched forever.

[image]

emilk, 2023-04-18:
Related:
* https://github.com/rerun-io/rerun/issues/1905

emilk, 2023-04-18:
There are many ways to solve this:
* GC the oldest recording first
* GC all at the same rate
* …

Different use cases have different requirements. What is obvious that the _current_ behavior sucks.

I suggest we just GC every open recording, OR drop the oldest recording, whatever is easier.

nikolausWest, 2023-09-11:
Related ?
- https://github.com/rerun-io/rerun/issues/1803

teh-cmc, 2023-11-07:
As far as I'm aware we now evenly distribute GC pass across all recordings:
```rust
pub fn purge_fraction_of_ram(&mut self, fraction_to_purge: f32) {
    re_tracing::profile_function!();

    for store_db in self.store_dbs.values_mut() {
        store_db.purge_fraction_of_ram(fraction_to_purge);
    }
}
```

which is overall an improvement, but there probably should be a blueprint setting to configure whether you want to distribute evenly, or prioritize cleaning up previous recordings of the same app_id first.

nikolausWest, 2023-11-07:
I think we should make the default behavior be to drop old data in a way that if you are running serial experiments, old recordings get dropped first, and if you are doing parallel experiments, we drop data from all recordings evenly. That is, we should drop based on time.

nikolausWest, 2023-11-07:
As it stands, I think this is a blocker for the hugging face spaces demo

emilk, 2023-11-07:
A good starting strategy: only run gc on oldest recording. When it is empty, close it.

Werinkle: row-protection

PR #4183 (2023-11-09): Implement recording/last-modified-at aware garbage collection
**Commit by commit, there's renaming involved!**

GC will now focus on the oldest-modified recording first.
Tried a lot of fancy things, but a lot of stress testing has shown that nothing worked as well as doing this the dumb way.

Speaking of stress testing, the scripts I've used are now committed in the repository. Make sure to try them out when modifying the GC code :grimacing:.

In general, the GC supports stress much better than I thought/hoped:
- `many_medium_sized_single_row_recordings.py`, `many_medium_sized_many_rows_record [... cut]
commit b8ea6af0 (2023-11-09): Implement recording/last-modified-at aware garbage collection (#4183)

**Commit by commit, there's renaming involved!**

GC will now focus on the oldest-modified recording first.
Tried a lot of fancy things, but a lot of stress testing has shown that
nothing worked as well as doing this the dumb way.

Speaking of stress testing, the scripts I've used are now committed in
the repository. Make sure to try them out when modifying the GC code
:grimacing:.

In general, the GC supports stress much better than I thought/hoped:
- `many_medium_sized_single_row_recordings.py`,
`many_medium_sized_many_ro [... cut]
closed 2023-11-09 by teh-cmc