rerun-io/rerun #1932: `multithreading` example dumps a stacktrace all while running to completion

teh-cmc, 2023-04-19:
```
$ ./examples/python/multithreading/main.py
Traceback (most recent call last):
  File "pyarrow/pandas-shim.pxi", line 110, in pyarrow.lib._PandasAPIShim._check_import
  File "pyarrow/pandas-shim.pxi", line 48, in pyarrow.lib._PandasAPIShim._import_pandas
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.10/site-packages/pyarrow/pandas_compat.py", line 24, in <module>
    import concurrent.futures.thread  # noqa
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 37, in <module>
    threading._register_atexit(_python_exit)
  File "/usr/lib/python3.10/threading.py", line 1504, in _register_atexit
    raise RuntimeError("can't register atexit after shutdown")
RuntimeError: can't register atexit after shutdown
Exception ignored in: 'pyarrow.lib._PandasAPIShim._have_pandas_internal'
Traceback (most recent call last):
  File "pyarrow/pandas-shim.pxi", line 110, in pyarrow.lib._PandasAPIShim._check_import
  File "pyarrow/pandas-shim.pxi", line 48, in pyarrow.lib._PandasAPIShim._import_pandas
  File "/home/cmc/dev/rerun-io/rerun/venv/lib/python3.10/site-packages/pyarrow/pandas_compat.py", line 24, in <module>
    import concurrent.futures.thread  # noqa
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 37, in <module>
    threading._register_atexit(_python_exit)
  File "/usr/lib/python3.10/threading.py", line 1504, in _register_atexit
    raise RuntimeError("can't register atexit after shutdown")
RuntimeError: can't register atexit after shutdown

$ echo $?
0
```

jleibs, 2023-04-19:
Does this happen deterministically for you? I can't repro locally.

jleibs, 2023-04-19:
Ok, a bit more digging on this... this *only* happens if (1) pandas is installed, and (2) we use pyarrow from inside of a thread.

Minimal repro without touching rerun:
```
import threading
import pyarrow

def use_pyarrow() -> None:
    table = pyarrow.table({"a": [1, 2, 3]})

def main() -> None:
    t = threading.Thread(target=use_pyarrow, args=())
    t.start()

if __name__ == "__main__":
    main()
```

Will create an issue on pyarrow github.

PR #1934 (2023-04-19): Join threads at end of multi-threading example
This is a generally reasonable thing to do and prevents the error from:
 - https://github.com/rerun-io/rerun/issues/1932

Fixes #1932 

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [ ] I've included a screenshot or gif (if applicable)
closed 2023-04-19 by teh-cmc